bblanchon/ArduinoJson #1144: ESP8266 and ESP32 crashing after succesful parsing

ElToberino, 2019-12-12:
I use ArduinoJson V6 with the current ESP8266 Core 2.6.2
The same problems also occurs with an ESP32.

I'm facing a strange issue with the function below I call during void setup();
Http-Connection is established and the parsing also works fine.
Calling the next function in setup, the ESP crashes and reboots with "Exception 9 (LoadStoreAlignmentCause)".

My function:
```
void getNews() {
  WiFiClient client;
  HTTPClient http;

   for (uint8_t w=0; w<newssources; w++){                    //loop for number of news sources
  
    String url;
    uint8_t newsmultiplier = numofarticles * w;
  
   if (w==0){
    url = "http://newsapi.org/v2/top-headlines?sources=spiegel-online&pageSize=3&page=1&apiKey=" + na_APIKEY;
  }
  if (w==1){
    url = "http://newsapi.org/v2/everything?domains=derstandard.at&pageSize=3&page=1&apiKey=" + na_APIKEY;
  }
  if (w==2){
    url = "http://newsapi.org/v2/everything?domains=tirol.orf.at&pageSize=3&page=1&apiKey=" + na_APIKEY;
  }

  if(http.begin(client,url)){    

   int httpCode = http.sendRequest("GET");

 
  if(httpCode == 200) {

   DynamicJsonDocument doc(5000);

    DeserializationError error = deserializeJson(doc, http.getStream());
    if (error) {
      Serial.println("deserializeJson() failed: ");
      Serial.println(error.c_str());
      return;
      }
  
   else { 
    JsonArray articles = doc["articles"];

    JsonObject articles_0 = articles[0];
    const char* name0 = articles_0["source"]["name"]; 
    const char* title0 = articles_0["title"]; 
    const char* url0 = articles_0["url"]; 

    JsonObject articles_1 = articles[1];
    const char* name1 = articles_1["source"]["name"];
    const char* title1 = articles_1["title"]; 
    const char* url1 = articles_1["url"]; 

    JsonObject articles_2 = articles[2];
    const char* name2 = articles_2["source"]["name"]; 
    const char* title2 = articles_2["title"]; 
    const char* url2 = articles_2["url"]; 
    
    

    snprintf(table[newsmultiplier].newscur, sizeof(table[newsmultiplier].newscur), "%s %s: %s", msgNews, name0, title0);
    snprintf(table[newsmultiplier].newslink, sizeof(table[newsmultiplier].newslink), "%s", url0);

    snprintf(table[newsmultiplier+1].newscur, sizeof(table[newsmultiplier+1].newscur), "%s %s: %s", msgNews, name1, title1);
    snprintf(table[newsmultiplier+1].newslink, sizeof(table[newsmultiplier+1].newslink), "%s", url1);

    snprintf(table[newsmultiplier+2].newscur, sizeof(table[newsmultiplier+2].newscur), "%s %s: %s", msgNews, name2, title2);
    snprintf(table[newsmultiplier+2].newslink, sizeof(table[newsmultiplier+2].newslink), "%s", url2);

    }
  }
  else {
   Serial.printf("http error: ", http.errorToString(httpCode).c_str());
   }

  Serial.println(table[newsmultiplier].newscur);
  Serial.println(table[newsmultiplier+1].newscur);
  Serial.println(table[newsmultiplier+2].newscur);
  Serial.println(table[newsmultiplier].newslink);
  Serial.println(table[newsmultiplier+1].newslink);
  Serial.println(table[newsmultiplier+2].newslink);

   }
   else {
    Serial.println("Unable to connect to news API"); 
   }
   http.end();
   }
   }
```

I used Json Assistant which calculated a doc size of about 2800.
I tried doc size with 3000, 4000, 5000 an higher, but that didn't change anything - ESP keeps crashing though parsing and printing into the arrays had worked!

I have no idea what happens - any ideas would be appreciated. Thank you!

bblanchon, 2019-12-13:
Hi @ElToberino,

I'm sorry, I don't see anything wrong with the snippet above.
What makes you think that ArduinoJson causes the crash?
What does [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) say?

Best Regards,
Benoit

ElToberino, 2019-12-13:
Thank you for your answer. I don't know what happens and why. Exception decoder shows the following:

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 11 results
0x402272ab: tcp_close at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c line 493
0x40214465: __delay at C:\Users\Isabel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/core_esp8266_wiring.cpp line 57
0x4020dd98: WiFiServer::close() at C:\Users\Isabel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WiFi\src/WiFiServer.cpp line 134
0x402065a1: setup at C:\Users\Isabel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 371
:  (inlined by) ?? at C:\Users\Isabel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 103
:  (inlined by) setup at C:\Users\Isabel\Documents\Arduino\Tobers Multidisplay\multidisplay_XXXXA/multidisplay_XXXXA.ino line 2284
0x4021e3b0: spiffs_cache_page_get_by_fd at ?? line ?
0x4021e3b0: spiffs_cache_page_get_by_fd at ?? line ?
0x4021e3b0: spiffs_cache_page_get_by_fd at ?? line ?
0x4021e3b0: spiffs_cache_page_get_by_fd at ?? line ?
0x4021e3b0: spiffs_cache_page_get_by_fd at ?? line ?
0x40213d58: loop_wrapper() at C:\Users\Isabel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/core_esp8266_main.cpp line 177
0x40100cd1: cont_wrapper at C:\Users\Isabel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/cont.S line 81

```

Best regards!

bblanchon, 2019-12-18:
Hi @ElToberino,

I don't think this problem is related to ArduinoJson.
However, if you believe ArduinoJson causes this exception and you want me to take a closer look, please write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) so I can reproduce the issue here.

Best Regards,
Benoit

ewaldc, 2019-12-23:
@ElToberino,
If the site returns a chunked response, than passing http.getStream() won't work as reading from the stream will not filter out the chunk headers which are not valid json.  In my case it also caused the ESP to crash during call to deserialize.   Also, with both httpclient code and ArduinoJson deserialize in one function, you might start to get crashes on the ESP8266 with JSON strings around 10K (depending on how much available RAM you have), since you have the JSON string multiple times in memory: in TCP stack, in HTTPClient buffers and in JsonDocument.   Specific unicode characters might also cause issues, normally _InvalidInput_ is returned as error, but in one particular case, I had the ESP crash as well.

To narrow down the issue, you might try:
- find out where in the code the crash occurs (e.g. is it in deserialize or after ?).
- capture the HTTP response first (e.g. to a SPIFFS file or to a string) and then deserialize.  This eliminates a series of issues with memory, TCP timeouts etc.   

```
SPIFFS.format();	//Without format, SPIFFS might cause resets...		
File wfile = SPIFFS.open(F("<filename of choice>"), "w");
httpClient.writeToStream(&wfile); // buffered write to stream (1460 bytes at the time)
wfile.close();
httpClient.end();		//Close connection free up TCP memory
// Now, deserialize from file or String.

Ewald

```

bblanchon, 2020-01-10:
Thanks for this message @ewaldc; I didn't realize that using `getStream()` bypasses the chunked transfer decoding. I updated [the documentation](https://arduinojson.org/v6/how-to/use-arduinojson-with-esp8266httpclient/) accordingly.

@ElToberino, did you make some progress?

ElToberino, 2020-01-15:
@ewaldc @bblanchon 
Thanks a lot for your answers and sorry for my late reply.
Though I don't know the reason, I'm sure that this crashes are caused by a memory problem as it was mentioned by @ewaldc. 
It's definitely not a parsing problem because parsing json and writing results into variables works fine - the crash occurs after that. Meanwhile I also doubt the problem is related to ArduinoJson, because I have a similar function (calling another API) working succesfully in the same sketch.
My workround is saving json into a string with http.getString() and parsing this string. I'd prefer parsing directly from stream, but this solution also works for me.
I'll continue testing and will report if I'm able to find the reason for the problem.

Thank you and best regards!

ElToberino, 2020-01-15:
Update:
Having tested some code variations this evening, I found out something interesting.
My code called functions in void setup() in the following order:
```
function_No.1()       // establishing WiFi Connection
function_No.2()      // Http Call and Json Parsing API 1 with ESP8266HTTPClient parsing from stream
function_No.3()      // Http Call and Json Parsing API 2 with ESP8266HTTPClient parsing from stream
function_No.4()      // server.begin()    with ESP8266WebServer
```
This makes ESP crashing. As I found out after a while the crash occurs on call of server.begin();
I changed the order of function calls to the following:
```
function_No.1()       // establishing WiFi Connection
function_No.2()      // server.begin()    with ESP8266WebServer
function_No.3()      // Http Call and Json Parsing API 1 with ESP8266HTTPClient parsing from stream
function_No.4()      // Http Call and Json Parsing API 2 with ESP8266HTTPClient parsing from stream
```
Surprisingly, this order doesn't cause any crash, everything works fine. I have no explanation at all for this behaviour, but I'm far from being an expert. Perhaps someone has an idea why this happens.

bblanchon, 2020-01-16:
This sounds like a use-after-free, depending on the order of operations, it can be unseen.
Or it could be as simple as a null-pointer-pointer-dereferencing caused by [memory fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html), which also depends on execution order.
Did you upload the code somewhere we can see it?
Can you try to [add traces](https://github.com/bblanchon/ArduinoTrace) to locate the line that causes the crash?

ewaldc, 2020-01-17:
@ElToberino,
Technically _ESP8266WebServer_ and _ESP8266HTTPClient_ are separate libraries. 
_server.begin()_ (typically) calls _WiFiServer.begin()_ which does nothing more than opening a TCP socket that listens and accepts connections on your chosen web server port.  So, without another issue, both sequences should be equal.
Hence, @bblanchon is right on IMHO.

bblanchon, 2020-01-28:
@ElToberino, did you make some progress?

ElToberino, 2020-01-31:
@bblanchon , thank you for answer! I'm still testing the new version of the code with the changed order of function calls and since I made these changes it is running succesfully!

Thanks to your and @ewaldc 's hints I read a lot about memory fragmentation, use after free ect.. That really helped me learning a lot understanding some of the basics you don't know as beginner who started his first programming attempts (sometimes luckily, sometimes not) with the simple copy and paste method from different sources. 

I'm still not completely sure what caused the crashes, but I can exclude it has to do anything with Arduino JSON. I tested the code above in different ways and also in other sketches and it works really fine.

My sketch whose primary function is driving a LED Matrix using the [MD-Parola library](https://github.com/MajicDesigns/MD_Parola) has become larger and larger after my first attempts because I was happy to get it running and added more and more functions - which makes the whole thing a little bit complicate. 

Once it's completely finished (which shouldn't be far from now) and all my test's are passed I will, of course, publish my project and the code.

Thank you very much for your helpfulness and for your great work! Without this it would be impossible for beginners like me to enter the (exciting) world of programming.

Best regards!

bblanchon, 2020-01-31:
You're very welcome, @ElToberino.
Thank you for your kind words.
Good luck with your project.

closed 2020-01-31 by bblanchon