bblanchon/ArduinoJson #1145: DynamicJsonDocument.clear() does not free memory

Aircoookie, 2019-12-14:
Hi Benoit,
as always huge thanks for the amazing library! After consulting your book and web doc, I'm still a bit stumped and I hope that you can help out.

I know that a `DynamicJsonDocument` releases its buffer once it goes out of scope. Now I have a use case where I need to free the memory pool before the DynamicJsonDocument goes out of scope to free up some memory (of course the document will not be used after clear).
According to [documentation](https://arduinojson.org/v6/api/jsondocument/clear/), the `clear()` method should release the memory, but it doesn't appear to do so.

Platform: ESP8266 Arduino Core 2.6.2 with Arduino IDE 1.8.9
ArduinoJson 6.13.0

MCVE:
```cpp
#include <ArduinoJson.h>

char testJsonString[] = "{\"text\":\"This is just an example object to be deserialized\"}";

void setup() {
  Serial.begin(115200);
  delay(2000);

  Serial.print("at boot: ");
  Serial.println(ESP.getFreeHeap());
  
  DynamicJsonDocument doc(8192);

  Serial.print("after allocation: ");
  Serial.println(ESP.getFreeHeap());
  
  deserializeJson(doc, testJsonString);
  JsonObject obj = doc.as<JsonObject>();
  
  const char* text = obj["text"];
  Serial.println(text);

  doc.clear();
  
  Serial.print("after clear: ");
  Serial.println(ESP.getFreeHeap());
}

void loop() {
  Serial.print("after setup: ");
  Serial.println(ESP.getFreeHeap());
  delay(5000);
}
```

Serial output:
```
at boot: 52008
after allocation: 43808
This is just an example object to be deserialized
after clear: 43808
after setup: 52008
```

As you can see, the available heap is not restored until setup() exits.
I apologize if this is user error, in that case please guide me in the right direction :)

 I also attempted to `delete &doc`, which causes a crash (Exception 3), presumably because the Document itself is not allocated on heap, just its buffer.

Happy holiday season!

Christian

bblanchon, 2019-12-16:
Hi Christian,

The documentation was a bit misleading, so I added this paragraph:

> This function resets the memory pool but doesn’t destroy it. The **memory pool is still present in memory** but is empty. If you want to release memory held by the memory pull, you must destroy the JsonDocument.

If you want to destroy a [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/), scope it.

Happy holiday season for you as well.
Benoit

Aircoookie, 2019-12-19:
Hi Benoit,

thank you for changing the documentation :)
I changed my structure to allocate the JsonDocument on a smaller scope, works like a charm!

Thanks!

closed 2019-12-19 by Aircoookie

bblanchon, 2019-12-19:
You're welcome @Aircoookie. Good luck with your project!