bblanchon/ArduinoJson #1146: [Doc v6] Behaviour of isNull() and is<const char*>() for null values

codefrey, 2019-12-16:
This is a suggestion for the improvement of the documentation regarding the handling of `null` in ArduinoJson v6.

Since it was not stated otherwise in the migration guide and elsewhere, I wrongly assumed that `JsonVariant::is<const char*>()` would still return `true` for a `null` value, as it was the case in v5. 

Furthermore, the difference between `success()` and `isNull()` should be highlighted in the migration guide, because `isNull()` also returns `true` if the key is present in the JSON but its value is `null`, which `success()` does not.

Lastly, there is a undocumented method `JsonVariant::isUndefined()`. Does this method have the same purpose as `success()` and/or is this method even supported?

bblanchon, 2019-12-17:
Hi @codefrey,

Thank you very much for these remarks. I'm sure they'll help other users.

I added the following sentences in the [migration guide](https://arduinojson.org/v6/doc/upgrade/):

> isNull() is not the exact opposite of success(): when the value is defined but is null, both isNull() and success() return true.
>
> Also, note that, in ArduinoJson 5, is<const char*>() returned true if the value was null; it’s no longer the case in version 6.

`isUndefined()` is only for internal use right now. As you probably saw, it's only defined on `JsonVariant`, and it's missing on all other types.

I was afraid that adding yet another method to the public interface would confuse users, and I was not sure anyone would call it. That's why `isUndefined()` is not officially supported.

What do you think?
Should I make `isUndefined()` part of the official API?
Do you need it for your project?

Best Regards,
Benoit

codefrey, 2019-12-18:
Thank you for enhancing the documentation.

Regarding `isUndefined()`:
I migrated the Homie project to ArduinoJson v6, whose JSON validation code now needs a bit of rework after I discovered the changed behaviour. The JSON validation does sometimes require a parameter to be provided, even though it may be null. For example, in the WiFi configuration the parameter `password` must be supplied, but will be `null`, if no password is needed for authentication. It is questionable whether this even makes sense or not, but nontheless this kind of check can not be performed with v6. So my vote would be for making this part of the API.

I think it would be a good idea to keep this method exclusive to `JsonVariant` to avoid confusion, like for example:

```
...
json["username"] = "testuser";
JsonObject user = json["username"];
user.isUndefined();
```

What should undefined return? `true`, because `username` is part of the JSON? Or `false`, because `username` was not defined as an object.

bblanchon, 2019-12-19:
With ArduinoJson 6, you can use [JsonObject::containsKey()](https://arduinojson.org/v6/api/jsonobject/containskey/) as an alternative to `isUndefined()`, but it requires an additional lookup.

If I had to include `isUndefined()` in the public API, I would put it on `JsonObject` as well, so as to be consistent and allow generic code. However, I'm not sure if it would be different from `JsonObject::isNull()`; ideally, I'd like it to behave like on `JsonVariant`, but I'm not sure I can.

codefrey, 2019-12-22:
You're right with `containsKey()`, I've fogotten about that. Possibly because of the documentation's recommendation to not use it. :)

For my own projects I've realized that treating a missing key the same as `null` is benificial anyways. I've created a pull request to Homie to remove the remaining instances of this case, so I personally don't need `isUndefined()` for my projects anymore. I still think it would be a good idea to introduce it, but I will leave the decision up to you.

closed 2019-12-22 by codefrey

bblanchon, 2019-12-22:
Hi @codefrey,

> treating a missing key the same as null is beneficial anyways

Can you elaborate on this?
I had the same intuition but I couldn't find a good example.

Best regards,
Benoit

codefrey, 2019-12-22:
It's kind of hard to elaborate, since it's also just my intuition.

But there are smaller config size possible, eg.
```
{
"ssid": "MyWiFiNetwork",
"password": null, // Open WiFi network
"ip": null, // Use DHCP
"gw": null, // Use DHCP
"mask": null // Use DHCP
}
```
can be reduced to
```
{
"ssid": "MyWiFiNetwork"
}
```
while still having the freedom of using both cases.

A practical benefit for me was when I wrote a recursive patching algorithm to apply a patch in the form of a JSON object over another JSON object. By treating the two cases equally, I can simply delete elements of the destination JSON by setting them to null in the patch JSON. Otherwise this algorithm would only be able to add keys to the destination JSON, but not to remove them, which would lead to a steadily growing JSON.

An added benefit of not treating `null` as `const char*` any more is that the `operator|` can now be used to specify a default value for strings, which can make checking the string for being a `nullptr` obsolete.

The only point I see in requiring `null` for a key is to make sure some logic did not forget about including this key in the JSON.