bblanchon/ArduinoJson #1147: "undefined reference" when passing JsonObject to function

nikolasdas, 2019-12-16:
Hey!
I'm trying to pass a JsonObject from a document to a static method.
`StaticJsonDocument<4096> json;
deserializeJson(json, payload, length);
EffectFactory::fromJson(json["effect"]);`

`static void EffectFactory::fromJson(const JsonObject& json) {
  ...
}`

Unfortunately I get an error compiling  the code:
`ArduinoJson/Variant/VariantData.hpp:184: undefined reference to EffectFactory::fromJson(ArduinoJson6130_100001::ObjectRef const&)'`

I tried using as<JsonObject> and to<JsonObject>, both const and not const parameter and as a reference and not... nothing seems to work. Any idea?

bblanchon, 2019-12-17:
Hi @nikolasdas,

This looks like a link error, not a compilation error: the linker cannot find the method.
Maybe you forgot to include the `.cpp` file in your build.
Or maybe it's not a static method of the `EffectFactory` class but a [static function](https://stackoverflow.com/q/558122/1164966) of `EffectFactory`  namespace.

BTW, using a const-reference is a good practice, but you can do even better.
In ArduinoJson 6, a `JsonObject` is already a "smart" reference, so you don't need to add another indirection. Moreover, there is a read-only version call `JsonObjectConst` (unfortunately undocumented right now):

    void EffectFactory::fromJson(JsonObjectConst json);

This will probably reduce the size of the executable a bit and slightly improve the performance.

Best Regards,
Benoit

nikolasdas, 2019-12-17:
Thank you! Found the problem... I was including the library in two files, but only declared a config variable (ARDUINOJSON_USE_LONG_LONG) in one of the files....

closed 2019-12-17 by nikolasdas