bblanchon/ArduinoJson #1148: DeseralizeJSON returns invalid input on equals strings

Yanchik0402, 2019-12-17:
Hello, I need to read a JSON from SD card, but this lib returns errors, then I decided to copy the output from the serial port and paste it directly into the code, but even then the library gives an error ... 
Maybe something in charset, but the characters are exactly the same ..

(code in file)

[testor.txt](https://github.com/bblanchon/ArduinoJson/files/3972482/testor.txt)

codefrey, 2019-12-22:
You have a unicode character ([byte order mark](https://en.wikipedia.org/wiki/Byte_order_mark) 0xEFBBBF) right before this backslash:
```
String test = "﻿{\"test\":\"10\"}"; //error
----------------^ here
```

bblanchon, 2019-12-22:
Thanks @codefrey.

@Yanchik0402, here is your find viewed with [HxD](https://mh-nexus.de/en/hxd/):

[image]

Yanchik0402, 2019-12-22:
In what charset I need to create files on SD so that there is no incompatibility?

codefrey, 2019-12-22:
ASCII or UTF-8 would be your best bets, as long as you are not using exotic characters. However, the encoding is not the problem with your file. The problem is that you have a character in the JSON which is not enclosed in quotation marks.
If we replaced the character causing the trouble (which is interpreted as a zero-width non-breaking space, so it's basically invisible) with an 'x', your JSON string would look like this:
`String test = "﻿{x\"test\":\"10\"}";`
which does not represent a valid JSON. Just remove this hidden character and you should be good to go.

closed 2020-01-09 by bblanchon