bblanchon/ArduinoJson #1150: True / False instead of 1 / 0

salvq, 2019-12-19:
Is this default intention to convert 1 / 0 to True / False ?

I am using simple reading digital input and serialization but output is True / False rather than 1 / 0. Is there kind of settings to change the output to 1 / 0?

Below are digital reads which are converted to true / false

**Code part**
```
  doc["dateStamp"] = getDateStamp();
  doc["timeStamp"] = getTimeStamp();
  doc["seconds"] = millis()/1000;
  doc["symptom"] = symptom;
  doc["msgId"] = messageCount;
  doc["mqttEr"] = mqttClient.connectError();
  doc["deviceId"] = deviceId;
  doc["yellow"] = !digitalRead(0);
  doc["green"] = !digitalRead(1);
  doc["cycle"] = !digitalRead(15);
  doc["red"] = !digitalRead(16);
  doc["reject"] = !digitalRead(17);
```
**Output**
```
14:27:32.668 -> {"dateStamp":"2019-12-19","timeStamp":"14:27:34","seconds":61,"symptom":"data","msgId":1,"mqttEr":0,"deviceId":"fautest","yellow":false,"green":false,"cycle":true,"red":true,"reject":true}
```

bblanchon, 2019-12-19:
Hi @salvq,

This is the expected behavior when you pass a boolean to ArduinoJson.
If you want 0 or 1, you need to cast the boolean to an integer:

```c++
doc["reject"] = (int)!digitalRead(17);
```

Best Regards,
Benoit

salvq, 2019-12-20:
Got it, thank you

closed 2019-12-20 by salvq