bblanchon/ArduinoJson #1151: [...] Brackets

Etyamor, 2019-12-20:
Hey, how to parse this kind of JSON files?
```
[
  { "name" : "Name" }
]
```
Only without this [] brackets it works.

Aircoookie, 2019-12-21:
Hi @Etyamor !
Those [] brackets are a JSON array, it may contain multiple objects or other data types.
See section 3.5 and 3.6 of the official documentation book: https://arduinojson.org/v6/doc/deserialization/

Here is a very simple example (partly copied from the book):
```cpp
// You need to know how big the JSON input can get
//it is helpful to use https://arduinojson.org/v6/assistant/
const int capacity = 1024;

// Allocate the JsonDocument
StaticJsonDocument<capacity> doc;

// Parse the JSON input
DeserializationError err = deserializeJson(doc, input);
if (err) return; //add error handling here

// Get the first object of array
JsonObject obj = doc[0];
const char* name = obj["name"];
```
I hope it works fine!

Have a happy holiday season :)

bblanchon, 2019-12-21:
In addition to the excellent answer from @Aircoookie, I'd say that you can paste your sample document in the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) so it'll show a sample deserialization program:

```c++
const char* json = "[{\"name\":\"Name\"}]";

deserializeJson(doc, json);

const char* root_0_name = doc[0]["name"]; // "Name"
```

Etyamor, 2019-12-23:
Thanks a lot guys.

closed 2019-12-23 by Etyamor