bblanchon/ArduinoJson #1152: JsonConfigFile Crash at compilation

BOBILLIER, 2019-12-20:


BOBILLIER, 2019-12-20:
It's strange' all my previous message seems to have disappeared. I try again.Perhaps too big?

I have just try to compile JsonConfigFile without any modification and i obtain error message:
call to 'fs::File SDClass::open(const char*, uint8_t)' uses the default argument for parameter 2, which is not yet defined

Esp8266 NodeMcu 1.0 / Arduino 1.8.10/ ArduinoJson 6.13.0

bblanchon, 2019-12-21:
Hi @BOBILLIER,

Your first message is in a comment (enclosed in ``), that's why it appears empty.

The first compilation error is:

```
In file included from <libraries>\SdFat\src/FatLib/FatLib.h:27:0,
                 from <libraries>\SdFat\src/SdFat.h:33,
                 from <packages>\esp8266\hardware\esp8266\2.6.3\libraries\SDFS\src/SDFS.h:36,
                 from <packages>\esp8266\hardware\esp8266\2.6.3\libraries\SD\src/SD.h:25,
                 from <libraries>\ArduinoJson\examples\JsonConfigFile\JsonConfigFile.ino:23:
<libraries>\SdFat\src/FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'

 class File : public FatFile, public Stream {
       ^
```

This error is not in the realm of ArduinoJson.
There is something wrong SD library.
Unfortunately, I don't have access to my computer right now, so I cannot try.

Best regards,
Benoit

closed 2020-01-03 by bblanchon