bblanchon/ArduinoJson #1158: How can i destroty a json document ?

GhassanYusuf, 2020-01-02:
I ran into a problem where my Arduino has several JSON documents created, even after clearing the object the memory is still full

can you show me how to destroy a JSON document after creating it. "I don't want to clear it I want to destroy it"

ewaldc, 2020-01-02:
@GhassanYusuf ,
According to the v6 documentation,  jsonDoc._clear()_ should release the memory, but in IMHO it does not free (all) the memory.  I have used two approaches with success:

- use a pointer to a JsonDocument -> release memory with _clear()_ and then _delete_ the pointer
- use JsonDocument in a separate function -> releases all memory when the JsonDocument goes out of scope

Maybe there is a way to call the JsonDocument destructor (if exposed), but I have not really searched for this or spend time on it.
Ewald

bblanchon, 2020-01-02:
Hi,

As stated in the documentation, [JsonDocument::clear()](https://arduinojson.org/v6/api/jsondocument/clear/) resets the memory pool but doesn't destroy it.

To destroy a `JsonDocument` and release all the associated memory, use the classic C++ destruction rules: either with a scope (preferred), with `new`/`delete`, or with a `unique_ptr`.

Best wishes,
Benoit

GhassanYusuf, 2020-01-02:
Please write me a short example

ewaldc, 2020-01-02:
On the ESP platform...
```
Serial.printf_P(PSTR("free heap memory: %d\n"), ESP.getFreeHeap());
DynamicJsonDocument doc(200);
deserializeJson(doc, "{\"name\": \"ArduinoJson\",\"stargazers\": {\"totalCount\": 3415},\"issues\": {\"totalCount\": 21}}");
Serial.printf_P(PSTR("free heap memory: %d\n"), ESP.getFreeHeap());
doc.~BasicJsonDocument();
Serial.printf_P(PSTR("free heap memory: %d\n"), ESP.getFreeHeap());
```

Produces:
free heap memory: 43960
free heap memory: 43752
free heap memory: 43960

If you put this code in a function, then there is no need for _doc.~BasicJsonDocument();_ as it will be invoked automatically at the end of the function.

bblanchon, 2020-01-03:
@GhassanYusuf, look how [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/) uses `JsonDocument`s as local variables; that's the best way to scope a `JsonDocument`.

GhassanYusuf, 2020-01-08:
`// CONVERTING GAME TEXT TO JSON
    DynamicJsonDocument   play(capacity); // building JSON Buffer
    
    // Deserialize the JSON document
    DeserializationError  error = deserializeJson(play, settings); // the main JSON body container

    // JSON ERROR
    if(error) {
      dualcomm(JSON_ERROR);
      return;
    }
    
    // PUBLIC VARIABLES
    uint16_t  limit         = play["lm"]; // Kick Target Limit
    bool      limitEnable   = play["le"]; // Enable Kick Target Limit

    // Clearing Buffer
    play.clear();

    // Trying To Delete The Document
    play.delete();`

I still don't get it

ewaldc, 2020-01-09:
Have you looked at the code that Benoit or I suggested?

GhassanYusuf, 2020-01-09:
> 
> 
> Have you looked at the code that Benoit or I suggested?

I have seen your example - I tested it, but what I need is to destroy the document which I mean destroy the dynamic JSON document object

i dont want it to exist after usage.

GhassanYusuf, 2020-01-09:
> @GhassanYusuf, look how JsonConfigFile.ino uses JsonDocuments as local variables; that's the best way to scope a JsonDocument.

[image]

None Of Them Showing How To Destroy, My Problem Is because of the document still exists, so clearing the document will not do.

> Hi,
>
> As stated in the documentation, JsonDocument::clear() resets the memory pool but doesn't destroy it.
>
> To destroy a JsonDocument and release all the associated memory, use the classic C++ [...]

Can you show me how C++ destruction rules apply on JsonDocument ? plz show an example code.

bblanchon, 2020-01-09:
Hi @GhassanYusuf,

I cannot find a better example than the one you posted above.

You don't see a line that destroys the `StaticJsonDocument` because C++ automatically destroys it.
Every variable is defined in a scope, and C++ destroys the variable when the instruction pointer leaves this scope.

In this case, `doc` is defined in the function scope, so it is destructed as soon as the function returns. The same goes for the `file` object, and it would be the same for an integer variable.

Does that make sense?

Best regards,
Benoit

ewaldc, 2020-01-09:
@GhassanYusuf ,
Benoit explained how in C++ a variable (JsonDocument in this case) is automatically destroyed when it goes out of scope at the end of a function.
In my example, I used _**doc.~BasicJsonDocument();**_ to delete the JsonDocument.
Every JsonDocumument is part of the BasicJsonDocument class, and ~BasicJsonDocument() is the destructor for this class (functions with **~** are C++ class destructors).

closed 2020-01-28 by bblanchon