bblanchon/ArduinoJson #1159: DeserializationError of ESP32 SPIFFS json file returns error 'incompleteInput'

400GreyBike, 2020-01-02:
I'm struggling to get a Json file in ESP32 SPIFFS to deserialize unless I create a buffer and read that.  In the SD example it shows 

> File file = SD.open(filename);  

and then calls 

> StaticJsonDocument<512> doc;
>   DeserializationError error = deserializeJson(doc, file);`

If I open my file using 

> File fileConfig = SPIFFS.open("/PM8Config.json", FILE_READ);   

and then deserialize using

> StaticJsonDocument<1000> doc;
> DeserializationError error = deserializeJson(doc, fileConfig);

I get a failure indicating "IncompleteInput"

If however I use the following, I am able to deserialise successfully where String data is in the json, but I get no data where int data items are i.e. pollRate

`File fileConfig = SPIFFS.open("/PM8Config.json", FILE_READ);                // Check if we can open the Config file 
    if (!fileConfig) {
        Serial.println("There was an error opening the Config file");
        Serial2.print(F("pgInitConfig.t10.txt=\"Failed"));
    //return false;
    }else{
        Serial.println("Config File was successfully opened");
        Serial2.print(F("pgInitConfig.t10.txt=\"Opened"));            
        //return true;
            size_t fileSize = fileConfig.size();
            std::unique_ptr<char[]> buf(new char[fileSize]);
            fileConfig.readBytes(buf.get(), fileSize);            
            StaticJsonDocument<1000> doc;
            DeserializationError error = deserializeJson(doc, buf.get());
            // Test if parsing succeeds.
            if (error) {
                Serial.print(F("deserializeJson() failed: "));
                Serial.println(error.c_str());
                //return;
            }
            int testJsonRead = doc["sensorData"]["pollRate"];
            Serial.print("Test JSON Read variable (should read '30': " );
            Serial.println(testJsonRead);
            const char* testPW = doc["password"];
            Serial.print("Test JSON Read variable (should show PW: " );
            Serial.println(testPW); `

The json file in SPIFFS is (and I can Serial.print this without an issue):
`{
    "platform": "NodeMCU32s",
    "framework": "Arduino",
    "version": "0.1a",
    "vdated": "30/12/19",
    "ssid": "myssid",
    "password": "mypassword", 
    "thingSpeakInfo":{
        "thingspeakAddress": "api.thingspeak.com",
        "thingspeakChannelID": 000000,
        "thingspeakReadApiKey": "myAPIKey",
        "thingspeakWriteApiKey": "myAPIKey"       
    }, 
    "checkWxApiKey": "myAPIKey",
    "sensorData":{
        "pollRate": 30,
        "stuff": "stuff"
    } 
}`

I'm clearly missing something here but can't figure out why I can't simply call the fileConfig object without having to use buf.get()?  

Any help would be appreciated and I think it's time to buy the book :-)

400GreyBike, 2020-01-02:
Having found another thread, it needs the removal of the readBytes() statement to ensure the pointer isn't at the end of the file.

Apologies for adding a somewhat wasted "issue" to an excellent library.

closed 2020-01-02 by 400GreyBike

bblanchon, 2020-01-03:
No worries @400GreyBike.
Thank you for using ArduinoJson.