bblanchon/ArduinoJson #1160: ArduinoJson parsing fail - Invalid Input

Nocturnalecode, 2020-01-06:
Hello, i recieve "Invalid Input" error always
Can you check my code if anything wrong?
Trying to debug it with prints but returning json seems correct and looks like this:
```
{
    "transmitterHost": "ESP_0EC076",
    "pairingCode": "625006a517cc4eb9b57d95a9654ba5e7"
}
```
When i print payload to serial port screen i see this line:

`{"transmitterHost":"ESP_0EC076","pairingCode":"625006a517cc4eb9b57d95a9654ba5e7"}
`
But parsing fails.

```
    String url = "http://"+host+"/Register/receiver?receiverHost="+hostStr;
    Serial.print("[HTTP] begin...\n");
    http.begin(url);  // HTTP
    Serial.println(url);

      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();
      Serial.println("[HTTP] GET DONE...\n");

          Serial.println("Trying to read http.getString()!");
          String payload = http.getString();
          Serial.println("String payload = http.getString() ok!");
          Serial.print("Incoming json: ");
          Serial.println(payload);

      std::string json = payload.c_str();

      Serial.println("");

      Serial.println("trying to create doc!");
      StaticJsonDocument<200> doc;
      Serial.println("DONE with create doc!");
      Serial.println("Trying deserializaJson...");
      DeserializationError error = deserializeJson(doc, json);

      if(error){
        Serial.print("deserialize basarisiz, kod: ");
        Serial.println(error.c_str());
        return;
      }
```
Also it's takes so long to get jsonstring (about 40 seconds). Using ESP8266HTTPClient library for that.
If problem is that, have any advice for getting json?
or if anythn else, have any idea?

Nocturnalecode, 2020-01-08:
I have figured that occurs on my office's internet connection but works good at home and figured something that json comes like this at home;

22:33:31.234 ->{"transmitterHost":"","pairingCode":"2366dfe831104da0aff6bb9769074fa8"}

but at office it comes like;

09:59:37.230 -> 47
09:59:37.230 -> {"transmitterHost":"","pairingCode":"2366dfe831104da0aff6bb9769074fa8"}
09:59:37.230 -> 0
09:59:37.230 -> 
09:59:37.230 -> 

have 47 before object, 0 and two whitespace after object.
is there a way to ignore those characters and just start from "{" and end at "}" ?

ewaldc, 2020-01-09:
@Nocturnalecode,
that is a HTTP 1.1 chunked response.  The _47_ means that the next chunk contains 47 bytes, and the _0_ means "end of transmit".  
But according to the ESP Arduino source code (from your code I assumed you are working with ESP 8266/32) on my system both _HTTPClient::getString()_ and _HTTPClient::writeToStream()_ should be able to recognize and handle these.

However it may be possible that I fixed a defect in the ESP Arduino _HTTPClient_ library and forget to post a PR.
Will go back and check and compare my version with the latest master.

Are you using the master branch on GitHub or a specific version?
Ewald

Nocturnalecode, 2020-01-09:
@ewaldc i am an alien for github actually trying to get used to. By changing connection it works well, not sure why that occurs on office's connection. Http getstring takes too long to get a response in office also. Might be rooter or service provider. Don't know.

bblanchon, 2020-01-09:
Hi @Nocturnalecode,

I agree with @ewaldc, ESP8266HTTPClient should handle [chunked transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding) for you. I don't know why it doesn't work; you should ask [the authors of the library](https://github.com/esp8266/Arduino).

By the way, I see that your program is duplicating the response several times. Please read [How to use ArduinoJson with ESP8266HTTPClient?](https://arduinojson.org/v6/faq/how-to-use-arduinojson-with-esp8266httpclient/) to learn how to improve your program.

Best regards,
Benoit

bblanchon, 2020-01-10:
Hi @Nocturnalecode,

It just came to my attention that ESP8266HTTPClient  doesn't handle chunked transfer encoding when we use `getStream()`, so you must call `useHTTP10(true)` before sending the request.
I updated [the documentation](https://arduinojson.org/v6/how-to/use-arduinojson-with-esp8266httpclient/) accordingly.

Best regards,
Benoit

Nocturnalecode, 2020-01-10:
@bblanchon @ewaldc thanks for help guys.

closed 2020-01-10 by Nocturnalecode

bblanchon, 2020-01-12:
You're welcome @Nocturnalecode. Thank you for using ArduinoJson.