bblanchon/ArduinoJson #1161: Json assistent SIZE shorten the JSON document

salvq, 2020-01-08:
Hello, just would like to highlight possible issue...

**According to arduino json assistant for the json below I should use SIZE(10)**
`{"symptom":"gsmUp","msgId":0,"deviceId":"test","yellow":0,"green":0,"cycle":0,"red":0,"reject":0,"dateCapture":"2020-01-08","timeCapture":"13:54:41"}
`

**When I use JSON_OBJECT_SIZE(10); result is not as desired**
```
{"symptom":"mqttUp","msgId":0,"deviceId":"test","yellow":0,"green":0,"cycle":0,"red":0,"reject":0,"dateCapture":null}
{"symptom":"data","msgId":1,"deviceId":"test","yellow":0,"green":0,"cycle":0,"red":0,"reject":0,"dateCapture":"2020-01-08"}
```

**When I use for example S JSON_OBJECT_SIZE(12); result is OK**
```
{"symptom":"mqttUp","msgId":0,"deviceId":"test","yellow":0,"green":0,"cycle":0,"red":0,"reject":0,"dateCapture":"2020-01-08","timeCapture":"13:54:56"}
{"symptom":"data","msgId":1,"deviceId":"test","yellow":0,"green":0,"cycle":0,"red":0,"reject":0,"dateCapture":"2020-01-08","timeCapture":"13:54:57"} 
```

**Faunction**
```
bool logSd(char* symptom) {
  char sdFilename[32];
  sprintf(sdFilename, "%02d%02d%02d%02d.csv", rtc.getYear(), rtc.getMonth(), rtc.getDay(), rtc.getHours());
  
  const size_t capacity = JSON_OBJECT_SIZE(10);
  DynamicJsonDocument doc(capacity);

  doc["symptom"] = symptom;
  doc["msgId"] = messageCount;
  doc["deviceId"] = deviceId;
  doc["yellow"] = Iono.read(DI5);
  doc["green"] = Iono.read(DI6);
  doc["cycle"] = Iono.read(DI4);
  doc["red"] = Iono.read(DI1);
  doc["reject"] = Iono.read(DI2);
  doc["dateCapture"] = getDateStamp();
  doc["timeCapture"] = getTimeStamp();

  file = SD.open(sdFilename, FILE_WRITE);

  WriteBufferingStream bufferedFile(file, 64);
  serializeJson(doc, bufferedFile);
  bufferedFile.flush();
  
  file.print("\n");
  file.close();
  return 1;
}
```

salvq, 2020-01-08:
@bblanchon Just upgraded to 6.13.0 and same issue...SIZE10 does not show complete JSON

bblanchon, 2020-01-09:
Hi @salvq,

You overlooked the second value returned by the assistant: "Additional bytes for strings duplication."

Indeed, when you pass a `Stream` to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/), ArduinoJson needs extra room to store the strings from the input.

You'll see that the "Parsing program" generated by the Assistant included these extra bytes.

Best regards,
Benoit

salvq, 2020-01-09:
Hi, I did not checked and have not add up additional bytes, was kind of expecting this is done in background 😊

Let me know of my understanding correct....

**Input**
`{"symptom":"data","msgId":1,"deviceId":"test","yellow":0,"green":0,"cycle":0,"red":0,"reject":0,"dateCapture":"2020-01-08","timeCapture":"13:54:57"}`

**My input to serializing process**
`const size_t capacity = JSON_OBJECT_SIZE(10) + 107;`

**Parsing assistant**
`const size_t capacity = JSON_OBJECT_SIZE(10) + 120;`
Why there is 120 in parsing program but memory pool size assistant says only additional 107 ?

bblanchon, 2020-01-09:
It's 120 instead of 107 because the assistant adds some margin, in case the input varies.
It's probably not enough slack though, so you may want to increase it a bit.

salvq, 2020-01-09:
Thank you sir, got it.

Last question...what value from assistant use for char payload size ? What is the best approach to calculate ? Any best practice ?

Best regards

`char payload[XXX];`

```
    const size_t capacity = JSON_OBJECT_SIZE(6);
    DynamicJsonDocument doc(capacity);

    doc["countfield"] = 1;
    doc["msgid"] = ++messageCount;
    doc["counttrig"] = "cycle";
    doc["redtag"] = Iono.read(DI1);
    doc["greentag"] = Iono.read(DI6);
    doc["yellowtag"] = Iono.read(DI5);

    char payload[258];
    size_t payloadSize = serializeJson(doc, payload);
```

bblanchon, 2020-01-10:
The payload size is the length of the serialized JSON document.
The Assistant doesn't print this value, but you can measure it with your text editor.
As a best practice, I recommend using a buffer slightly larger than the longest expected document.

bblanchon, 2020-01-10:
PS: I modified the Assistant so that it shows the length of the input

salvq, 2020-01-10:
This is just great, much easier than using word count function :)

Very appreciate your quick response !

👍

closed 2020-01-10 by salvq