bblanchon/ArduinoJson #1164: how big the capacity of DynamicJsonDocument can be for esp32

ligantx, 2020-01-09:
i use an esp32-cam for my project, and i want to parse a big json file from the client response.
So, in order to calculate a possible capacity i took a big json response and duplicate it in arduinoJson assistance. 
It gave me: const size_t capacity = JSON_ARRAY_SIZE(382) + 382 * JSON_OBJECT_SIZE(8) + 64440;

When i use it (even with much smaller json) i get "no memory" error.

When i change to the "real" capacity of my current json response to:
const size_t capacity = JSON_ARRAY_SIZE(30) + 30*JSON_OBJECT_SIZE(8) + 4880;

it works fine..

So my questions are:
1) why i get this error even i give so much space? is there any limit for capacity in esp32 
2) i guess deserializeJson(doc, client) is much more efficient in speed and memory compared to get the payload String and do deserializeJson(doc, payload)..am i correct?

bblanchon, 2020-01-09:
Hi @ligantx,

If you try to allocate a [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) that is too large to fit in memory, the allocation fails and [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) returns [NoMemory](https://arduinojson.org/v6/api/misc/deserializationerror/).

The ESP32 has 520KB of RAM, but many boards extend this capacity with an external PSRAM chip. Assuming your board embeds such a chip, your program must use dedicated allocation functions to use it.

To use the external PSRAM with ArduinoJson, you must use a custom allocator. Here is an example from the documentation of [BasicJsonDocument<T>](https://arduinojson.org/v6/api/basicjsondocument/):

```c++
struct SpiRamAllocator {
  void* allocate(size_t size) {
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
  }
  void deallocate(void* pointer) {
    heap_caps_free(pointer);
  }
};

using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
```

`SpiRamJsonDocument` is similar to [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/), except that it uses the external PSRAM.

Best regards,
Benoit

bblanchon, 2020-01-10:
PS: I added [How to use external RAM on an ESP32?](https://arduinojson.org/v6/how-to/use-external-ram-on-esp32/) to the documentation.

ligantx, 2020-01-10:
actually this is much better than a simple answer!

question: so the allocation that i tried `const size_t capacity = JSON_ARRAY_SIZE(382) + 382 * JSON_OBJECT_SIZE(8) + 64440;` was larger even than the 520KB of esp32 ram?

bblanchon, 2020-01-12:
I don't know the value, you have to check with the assistant, but it's most likely larger than 520KB.

`no matching function for call to 'BasicJsonDocument<SpiRamAllocator>::BasicJsonDocument()'` means you forgot to pass the capacity. I'm sorry, I forgot to put it in the example:

```c++
SpiRamJsonDocument doc(1048576);
deserializeJson(doc, input);
```

closed 2020-01-28 by bblanchon