bblanchon/ArduinoJson #1165: How to debug data coming from Serial when deserializeJson(doc, Serial) is used)?

mburakerbay, 2020-01-13:
Hi,
I am using ArduinoJson to communicate on arduino and esp8266.
Arduino is sending data to esp using serializeJson(saveDataDoc, EspSerial);
Esp is using deserializeJson(doc, Serial) to get data.
I want to debug/look json data which creates DeserializationError when esp try to deserialize data came from Arduino Serial. So i can understand whats wrong in data.

bblanchon, 2020-01-13:
Hi @mburakerbay,

Thank you for this question.
I answered with a new page in the documentation: [How to debug data coming from a stream?](https://arduinojson.org/v6/how-to/debug-data-coming-from-a-stream/).

Best regards,
Benoit

mburakerbay, 2020-01-13:
@bblanchon  Thanks for new how to documentation and fast replay. I will try it asap.

mburakerbay, 2020-01-13:
@bblanchon 
In arduino part i can use `ReadLoggingStream` to logging data because there are two serial. First one is Arduino's serial and second one is Esp's softserial. But in esp part there is only one serial so i can not use like `ReadLoggingStream loggingStream(EspSerial, Serial);`. How can i use if there is one serial?

bblanchon, 2020-01-14:
@mburakerbay, you could log to a file.
How do you usually debug your app?

mburakerbay, 2020-01-18:
Sorry for being late.
When i want to debug only esp, i am pluging only esp to computer and writing to Serial everything. While working with arduino, i am posting to local website :smiley: 
My problem is beginning to here, i am using httpclient to post like `serializeJson(jo, postData);` and getting payload like `deserializeJson(doc, http.getStream());`. I want to log what i serialized and deserialized, obviously i can not use `ReadLoggingStream` because i have only one `Serial` in esp.

bblanchon, 2020-01-20:
@mburakerbay, `ReadLoggingStream` is not limited to `Serial`.
It works with any implementation of `Print`, such as `WiFiClient` or [`StringPrint`](https://github.com/bblanchon/ArduinoStreamUtils#writing-to-a-string).

closed 2020-01-28 by bblanchon