bblanchon/ArduinoJson #1167: Moving json object within JsonDocument

Gei0r, 2020-01-13:
I have a JsonDocument which contains the following structure:

```
{ "a": "Hi there!", "b": 1337}
```
Now I want to transform this JsonDocument so that it contains:
```
{
    "additionalInfo": 4711,
    "original": { "a": "Hi there!", "b": 1337}
}
```

(In reality the "original" structure is larger. The JsonDocument has enough space for the change.)

I've read through the API documentation but I couldn't find a way to get this to work without allocating a new JsonDocument. 

Alternatively, I could create the original JsonObject as a "free-floating" object (not attached to a document root), but I don't think that's possible with v6?

bblanchon, 2020-01-14:
Hi @Gei0r,

I'm sorry, this was possible with ArduinoJson 5, but with ArduinoJson 6, you need to create a new `JsonDocument`.

I realize that this answer might disappoint you, but you have a very uncommon use case.
With version 6, I decided to simplify the API, sometimes sacrificing edge cases like this one.
I'm sure you understand.

Best regards,
Benoit

Gei0r, 2020-01-14:
No problem, thanks for the clarification.

closed 2020-01-14 by Gei0r