bblanchon/ArduinoJson #1168: no matching function for call to 'ArduinoJson6130_000001::BasicJsonDocument<ArduinoJson6130_000001::DefaultAllocator>::BasicJsonDocument()'

BritneyMuller, 2020-01-14:
Watched your video on how to upgrade from ArduinoJson 5 to 6.

Tried to implement your changes but know I'm missing something (this is for a [Paper Signals](https://papersignals.withgoogle.com/) project):

`
#include "APICalls.h"

// #define PRINT_PAYLOAD

String urlencode(String str)
{
    String encodedString="";
    char c;
    char code0;
    char code1;
    char code2;
    for (int i =0; i < str.length(); i++){
      c=str.charAt(i);
      if (c == ' '){
        encodedString+= '+';
      } else if (isalnum(c)){
        encodedString+=c;
      } else{
        code1=(c & 0xf)+'0';
        if ((c & 0xf) >9){
            code1=(c & 0xf) - 10 + 'A';
        }
        c=(c>>4)&0xf;
        code0=c+'0';
        if (c > 9){
            code0=c - 10 + 'A';
        }
        code2='\0';
        encodedString+='%';
        encodedString+=code0;
        encodedString+=code1;
        //encodedString+=code2;
      }
      yield();
    }
    return encodedString;
}

int days_in_month[13] = {0,31,28,31,30,31,30,31,31,30,31,30,31};

int leap_year(int year) {
    if(year%400==0) return 1;

    if(year%4==0 && year%100!=0) return 1;

    return 0;
}

int number_of_days(int year, int day, int month) {
    int result = 0;
    int i;

    for(i=1; i < year; i++) {
        if(leap_year(i))
            result += 366;
        else
            result += 365;
    }

    for(i=1; i < month; i++) {
        result += days_in_month[i];

        if(leap_year(year) && i == 2) result++;
    }

[... 722 lines cut ...]

        // file found at server
        if(httpCode == HTTP_CODE_OK) {
            payload = http.getString();
        }
    } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }

    http.end();

    #ifdef PRINT_PAYLOAD
        Serial.println("reply was:");
        Serial.println("==========");
        Serial.println(payload);
        Serial.println("==========");
    #endif
    return payload;
}
void PaperSignals::RunPaperSignals()
{
  String JSONData = getSignalByID(SignalID.c_str());
  ParseIntentName(currentIntent, JSONData);
}
`

ewaldc, 2020-01-14:
@BritneyMuller,
In ArduinoJson v6 there is no _JsonDocument.parseOject()_ call
So instead of e.g.
>      JsonObject& root = doc.parseObject(payload);
use
> DeserializationError err = deserializeJson(doc, payload);
> if (error) ... // check for parsing error here
> JsonObject root = doc.as\<JsonObject>(); // get the root object

There are some nice examples in the documentation...
Hope this helps, 
Ewald

bblanchon, 2020-01-14:
Hi @BritneyMuller,

I added a new page in the documentation to cover this error: [no matching function for call to 'BasicJsonDocument::BasicJsonDocument()'](https://arduinojson.org/v6/error/no-matching-function-for-call-to-basicjsondocument-basicjsondocument/).

In short, the solution is to pass an argument to the constructor of `DynamicJsonDocument`:

```c++
DynamicJsonDocument doc(capacity);
```

To compute the capacity, use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/), or read [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/).

Of course, what @ewaldc said applies as well.

Best regards,
Benoit

BritneyMuller, 2020-01-14:
Thank you both so much! That definitely helped!

--Ended up rolling back the installed ArduinoJson version to get my code uploaded per: https://github.com/googlecreativelab/paper-signals/issues/27

closed 2020-01-14 by BritneyMuller