bblanchon/ArduinoJson #1169: v5-v6 migration strcmp() causes crash

yknivag, 2020-01-16:
Trying to convert an ESP8266 sketch from ArduinoJSON v5 to v6.13.0

In v5 after `const char* pre_release = root["prerelease"];` I was able to use `if (strcmp(pre_release, "true") == 0)` to check if that key held the string true and if worked fine.

But in v6 after `const char* pre_release = doc["prerelease"];` the line `if (strcmp(pre_release, "true") == 0)` causes `Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads` and the uC restarts.

It compiles perfectly well against either v5 or v6 of the library but when using v6 the `strcmp()` causes a crash.

The documentation (and previous issues raised on here) suggest that `strcmp()` is the correct approach, I just can't get it to work.

Any ideas why the change of version should cause this?

yknivag, 2020-01-16:
Solved the immediate issue by replacing `const char* pre_release = doc["prerelease"];` with `bool pre_release = doc["prerelease"];`

closed 2020-01-16 by yknivag

bblanchon, 2020-01-17:
Hi @yknivag,

If "prerelease" is a string (and not a boolean), you can write:

```c++
if (doc["prerelease"] == "true") {

}
```

See a demo here:
https://wandbox.org/permlink/9edchNsphmH8icAW

Best regards,
Benoit

yknivag, 2020-01-17:
Hi Benoit,

Thank you for responding, especially as I'd closed the ticket.

`prerelease` was a `const char*` containing either the text "true" or "false".

I have fixed it, after further analysis using your excellent automated assistant by parsing the field directly into a boolean.

I was rather confused though that using `strcmp()` worked under v5 but not under v6. Maybe out of memory?

Probably my misunderstanding somewhere, was just confusing.

Thanks again for getting back.

bblanchon, 2020-01-17:
ArduinoJson parser was lazy: everything was stored as a string and only converted when the user asked.
In this case, it stored the string `"true"`, so when you asked a `const char*`, it gave you this string.

ArduinoJson 6, however, has an eager parse, it stores each value with the corresponding type. In this case, it stores a `bool`, so when you ask for a `const char*`, it cannot return a string and therefore returns NULL.

yknivag, 2020-01-17:
Once again, many thanks for getting back.

That makes perfect sense!