bblanchon/ArduinoJson #1170: create nested array of unsigned integers

TLS1000, 2020-01-20:
Hi, first of all thanks for making this library. I'm planning on it to send ~120 values to AWS.

I would like to be able to tell the serialization that values should be treated as 'unsigned int'. How can I avoid that they are treated as 'long'?

what I would like to do:

```

StaticJsonDocument<1098> jsonDoc;
  JsonObject stateObj = jsonDoc.createNestedObject("state");
  JsonObject reportedObj = stateObj.createNestedObject("reported");
  JsonArray negCellV = reportedObj.createNestedArray("negCellV");
  for(int i=0;i<6;i++){
    negCellV.add(unsigned int(50000));
  }

```

bblanchon, 2020-01-20:
Hi @TLS1000,

The correct syntax would be `(unsigned int)50000` or `static_cast<unsigned int>(50000)`.
But, what do you mean by "they are treated as long"?

Best Regards,
Benoit

TLS1000, 2020-01-21:
Hi Benoit, thanks for your response and great work on this library. I experimented with the hornbill library as well, it also works but I prefer yours because you're using more of the native esp32-arduino libraries, and there's activity on this library.

About the unsigned integers, in fact I don't know how they are treated... I need to set the array size via this 'JSON_ARRAY_SIZE(120)'. but how many bytes are allocated? how does the library determine if it's an array of uint8_t or an array of 'long' type?  

Something else: I'm using your code example, slightly modified with the addition of 120 uint16_t values. Where your example code ran for several days, after adding the 120 values (and send interval of 10s), the connection to AWS is lost after a few hours and the program doesn't seem to be aware there's anything wrong; No reconnect attemps or so... Maybe we need to check the timestamp of the shadow document after sending a message, and if there's a problem stop the client and reconnect?

bblanchon, 2020-01-21:
Hi @TLS1000,

> how many bytes are allocated?

The result of `JSON_ARRAY_SIZE(n)` is platform dependant.
On ESP32, it should be 16xn.
So, for n=120, it should be about 2KB.

> how does the library determine if it's an array of uint8_t or an array of 'long' type?

ArduinoJson uses template type deduction to determine the type of the value.
Most of the time, you don't have to worry about it.

> the connection to AWS is lost after a few hours

I'm assuming that you use a "keep-alive" HTTP connection.
This is a non-standard feature, so nothing forbids the server from closing the connection after a certain number of requests. You need to handle reconnection in your program.
Also, it could be a problem with the WiFi connection.

Best Regards,
Benoit

closed 2020-02-12 by bblanchon