bblanchon/ArduinoJson #1171: Unsigned long serialization/deserialization problem

hdimon, 2020-01-21:
Hi,

Sorry for disturbing but I've already spent a lot of time and unfortunately could not determine exact problem myself. I have only some guesses but all my proposed fixes don't work.

I have scetch running on Mega2560. Two controllers have the same codebase, but one of them sends serialized data to another via Serial1 port and other controller receives them via Serial2. Below there is simplified code. It's still very big though because I have to show some pieces of code since I don't know what exactly a reason of my issue.

```
Adafruit_BME280 bme;
float BME280Temperature = -1000;
float BME280Pressure = -100000;
float BME280Humidity = -1000;

unsigned long senseAirCO2PPM = 0;

void setup() {
  Serial.begin(9600);
  while(!Serial); // time to get serial running

  Serial.println(F("Initializing Serial1 (18, 19 pins)..."));
  Serial1.begin(9600);
  while(!Serial1); // time to get serial running

  Serial.println(F("Initializing Serial2 (16, 17 pins)..."));
  Serial2.begin(9600);
  while(!Serial2); // time to get serial running 
  
  senseAirSerial.begin(9600);
  while(!senseAirSerial);

  setupBME280();
    
  senseAirThread.onRun(SenseAirWorker);
  senseAirThread.setInterval(0);

  BME280Thread.onRun(BME280Worker);
  BME280Thread.setInterval(0);

  Serial1SendingDataThread.onRun(Serial1SendingDataWorker);
  Serial1SendingDataThread.setInterval(0);

  Serial2GettingDataThread.onRun(Serial2GettingDataWorker);
  Serial2GettingDataThread.setInterval(0);
}

void loop() {     
  if (senseAirThread.shouldRun())
     senseAirThread.run();

  if (BME280Thread.shouldRun())
     BME280Thread.run();

  if (Serial1SendingDataThread.shouldRun())
     Serial1SendingDataThread.run();

  if (Serial2GettingDataThread.shouldRun())
     Serial2GettingDataThread.run();
}

DynamicJsonDocument CollectJsonData() {
  DynamicJsonDocument doc(512);

  doc[F("CO2ppm")] = (unsigned int) senseAirCO2PPM;
  doc[F("TemperatureC")] = BME280Temperature;
  doc[F("PressureHPa")] = BME280Pressure / 100.0F;
  doc[F("HumidityPercent")] = BME280Humidity;

  return doc;
}

//SenseAirWorker code
void SenseAirWorker() { 
  switch (senseAirCurrentState) {
    case senseAir_WAIT_REQUEST_STATE:
      SenseAir_WaitRequest();
      break;
    case senseAir_SEND_REQUEST_STATE:
      SenseAir_SendRequest();
      break;
    case senseAir_RECEIVE_RESPONSE_STATE:
      SenseAir_ReceiveResponse();
      break;
  }
}

void SenseAir_ReceiveResponse() {
  if (senseAirSerial.available() == 0) {
    senseAirCurrentState = senseAir_WAIT_REQUEST_STATE;
    return;
  }
  
  for (int i = 0; i < 7; i++) {
    senseAirResponseBytes[i] = senseAirSerial.read();
  }

  unsigned int high = senseAirResponseBytes[3];
  unsigned int low = senseAirResponseBytes[4];
  senseAirCO2PPM = high * 256 + low;
  Serial.print("CO2 = ");
  Serial.println(senseAirCO2PPM);

  senseAirCurrentState = senseAir_WAIT_REQUEST_STATE;
}
//SenseAirWorker code

//BME280Worker code
void BME280Worker() {
  BME280_ReadValues();
}

void BME280_ReadValues() {
  BME280Temperature = bme.readTemperature();
  BME280Pressure = bme.readPressure();
  BME280Altitude = bme.readAltitude(SEALEVELPRESSURE_HPA);
  BME280Humidity = bme.readHumidity();
}
//BME280Worker code

//Serial1SendingDataWorker code
void Serial1SendingDataWorker() { 
  switch (serial1SendingDataCurrentState) {
    case serial1SendingData_GET_REQUEST_STATE:
      Serial1SendingData_GetRequest();
      break;
    case serial1SendingData_SEND_RESPONSE_STATE:
      Serial1SendingData_SendResponse();
      break;
  }
}

void Serial1SendingData_GetRequest() {
  while (Serial1.available()) {
    char inChar = (char)Serial1.read();

    if (inChar != '\n')
      serial1SendingDataInputCommanString += inChar;

    if (inChar == '\n' && serial1SendingDataInputCommanString.length() > 0) {
      serial1SendingDataInputCommanStringComplete = true;
      Serial.print(F("Serial1SendingData_GetRequest: received command = "));
      Serial.println(serial1SendingDataInputCommanString);
      
      serial1SendingDataCurrentState = serial1SendingData_SEND_RESPONSE_STATE;
      return;
    }
  }

  if (serial1SendingDataInputCommanStringComplete)
    serial1SendingDataCurrentState = serial1SendingData_SEND_RESPONSE_STATE;
}

void Serial1SendingData_SendResponse() {
  Serial.println(F("Serial1SendingData_SendResponse"));

  if (serial1SendingDataInputCommanString == F("get")) {
    DynamicJsonDocument doc = CollectJsonData();

    WriteBufferingStream bufferedFile(Serial1, 256);
    WriteLoggingStream loggingStream(bufferedFile, Serial);
    serializeJson(doc, loggingStream);
  }

  serial1SendingDataInputCommanString = "";
  serial1SendingDataInputCommanStringComplete = false;
  
  serial1SendingDataCurrentState = serial1SendingData_GET_REQUEST_STATE;
}
//Serial1SendingDataWorker code

//Serial2GettingDataWorker code
void Serial2GettingDataWorker() { 
  switch (serial2GettingDataCurrentState) {
    case serial2GettingData_WAIT_REQUEST_STATE:
      Serial2GettingData_WaitRequest();
      break;
    case serial2GettingData_SEND_REQUEST_STATE:
      Serial2GettingData_SendRequest();
      break;
    case serial2GettingData_RECEIVE_RESPONSE_STATE:
      Serial2GettingData_ReceiveResponse();
      break;
  }
}

void Serial2GettingData_WaitRequest() {
  if (!serial2GettingDataSendWaitRequestInterval.isRunning())
    serial2GettingDataSendWaitRequestInterval.start(serial2GettingDataSendWaitRequestIntervalMS);

  if (serial2GettingDataSendWaitRequestInterval.justFinished()){
    serial2GettingDataCurrentState = serial2GettingData_SEND_REQUEST_STATE;
    return;
  }
}

void Serial2GettingData_SendRequest() {
  bool finished = serial2GettingDataSendRequestDelay.justFinished();

  if (serial2GettingDataSendRequestDelay.isRunning() && !finished)
      return;

  if (finished) {
    serial2GettingDataCurrentState = serial2GettingData_RECEIVE_RESPONSE_STATE;
    return;
  }
      
  if (!Serial2.available()){
    Serial2.write("get\n");
    serial2GettingDataSendRequestDelay.start(serial2GettingDataSendRequestDelayMS);
    return;
  }

  serial2GettingDataCurrentState = serial2GettingData_RECEIVE_RESPONSE_STATE;
}

void Serial2GettingData_ReceiveResponse() {
  if (Serial2.available()) {
    ReadBufferingStream bufferingStream(Serial2, 256);
    ReadLoggingStream loggingStream(bufferingStream, Serial);
    deserializeJson(serial2GettingDataDoc, loggingStream);
  }

  Serial2GettingData_ProcessResponse();
}

void Serial2GettingData_ProcessResponse() {
  serial2GettingDataCurrentState = serial2GettingData_WAIT_REQUEST_STATE;
}
//Serial2GettingDataWorker code
```

Serializing object looks good (as I can see in serial monitor), for example:
`{"CO2ppm":528,"TemperatureC":23.58,"PressureHPa":994.4897,"HumidityPercent":36.08691}`
But deserializing string contains corrupted CO2 value most often it looks like:
`{"CO2ppm":10000,"TemperatureC":23.58,"PressureHPa":994.4897,"HumidityPercent":36.08691}`
though I saw not only 10000 but something like 2047 or 8314 and so on.., But last 2 days I see only 10000. I tried other ways to fix:). For example I did:
`doc[F("CO2ppm")] = (float) senseAirCO2PPM;`
I tried to extract code from CollectJsonData() and insert it directly in method Serial1SendingData_SendResponse().
I even tried to make string from CO2 parameter and add text symbol there to get something like:
`{"CO2ppm":"528test","TemperatureC":23.58,"PressureHPa":994.4897,"HumidityPercent":36.08691}`
but then I got:
`{"CO2ppm":"10000test","TemperatureC":23.58,"PressureHPa":994.4897,"HumidityPercent":36.08691}`
!!!
All of these facts leads me to idea that maybe something is wrong with my way of assigning senseAirCO2PPM variable. I undestand that sending data via uart takes some time and maybe bytes in memory are already changed by moment when serialized string is really sent to port. If I hardcode some value instead of 
`senseAirCO2PPM = high * 256 + low;`
everything works perfect. But I don't have idea how to fix that because firstly it's not clear for me why
`doc[F("CO2ppm")] = (unsigned int) senseAirCO2PPM;`
or
`doc[F("CO2ppm")] = (float) senseAirCO2PPM;`
does not work. In my understanding it should create new physical variable and it should not be corrupted...
Additionally I don't understand why it works for Temperature but does not work for CO2 though these are similar workers with similar code.
The more, serializarion works perfect with the same code when I send data via HTTP (I missed this code here because it does not make sense I think).

I really appreciate any help and ideas how to fix that or how to troubleshoot deeper.

Thanks in advance!

ewaldc, 2020-01-21:
@hdimon,
What do you get when you serialize the doc, then deserialize it and serialize it back on the same device?   If the final Json string is the same, then the problem is not with ArduinoJson since transmit serialized JSON, but somewhere in transmission (XON/XOFF?) or caused by data corruption.

hdimon, 2020-01-21:
@ewaldc,
Is it correct code for testing?

String jsonStr;
serializeJson(doc, jsonStr);
Serial.println(jsonStr);

DynamicJsonDocument doc1(512);
deserializeJson(doc1, jsonStr);
Serial.println(doc1.as<String>());

If yes, then it gives correct result...

bblanchon, 2020-01-21:
Hi @hdimon,

Can you confirm the following:

1. `{"CO2ppm":528,"TemperatureC":23.58,"PressureHPa":994.4897,"HumidityPercent":36.08691}` is printed by the `WriteLoggingStream` in `Serial1SendingData_SendResponse`
2. `{"CO2ppm":10000,"TemperatureC":23.58,"PressureHPa":994.4897,"HumidityPercent":36.08691}` is printed by the `ReadLogginStream` in `Serial2GettingData_ReceiveResponse`

If so, it looks like a transmission issue, which is highly unlikely at 9600 bauds.
Did you try to hardcode the string?

Also, I see that your code mentions threads in the `setup()` function.
Could it be a concurrency issue?
Can you try to declare `senseAirCO2PPM` as `volatile`?

Best Regards,
Benoit

hdimon, 2020-01-21:
Hi @bblanchon,

> Can you confirm the following:

Exactly.

> Did you try to hardcode the string?

Yes, it works...

> Could it be a concurrency issue?
> Can you try to declare senseAirCO2PPM as volatile?

I guess yes. The reason why I ask a question here because I totally don't understand where to find mistake - only in my code or in working JsonArduino with Serial.
I've just tried to do as you adviced:
1. Made `volatile unsigned long senseAirCO2PPM = 0;`
2. When form json doc in CollectJsonData() method I've made
```
 noInterrupts();
 int test = (int) senseAirCO2PPM;
 interrupts();
 doc[F("CO2ppm")] = test;
```
And still 10000 on receiving side...

bblanchon, 2020-01-22:
Hi @hdimon,

We need to reduce the amount of code to isolate the problem.
Remove every line of code that doesn't affect the outcome.
If it's a transmission problem, we should be able to reproduce it without ArduinoJson.
If it's a serialization problem, we should be able to reproduce it without the serial ports.
If it's a concurrency issue, it should disappear when we remove the concurrency.
It could also be a dangling pointer, but I don't think so because you said you always get the same value (1000).

Good luck.

Best regards,
Benoit

hdimon, 2020-01-22:
Hi @bblanchon,

I agree and I did that. And now I'm totally sure that it's not problem of ArduinoJson.
The deal is I reduced code just to
```
noInterrupts();
int test = (int) senseAirCO2PPM;
String jsonStr = String(test);

for (int i = 0; i < jsonStr.length(); i++)
  {
    Serial1.print(jsonStr[i]);   // Push each char 1 by 1 on each loop pass
    Serial.println(jsonStr[i]);
  }
interrupts();
```
And it still was reproduced! And I saw that Serial monitor shows correct chars on sending side but still gets wrong chars ("10000" or 49-48-48-48-48 in ascii code) on receiving side. I event tried to so such thing:
Declare
`volatile unsigned long senseAirCO2PPM = 0;`
And then in code of getting data from sensor
```
if (senseAirCO2PPM == 0)
  senseAirCO2PPM = high * 256 + low;
```
So it led to setting value only once during controller work to check if it's concurrency or not. And I saw then that it was enough to break things (First second receiving controller got 0 but then it became to get 10000 again). Thus I guess it's not about concurrency otherwise it should work.

I say "was" because already at night I decided to get back code but with keeping last changes with volatile. So code became looks like:
In declaring:
`volatile unsigned long senseAirCO2PPM = 0;`

In collect data method:
```
DynamicJsonDocument CollectJsonData() {
  DynamicJsonDocument doc(512);

  doc[F("Mac")] = macString;

  noInterrupts();
  int test = (int) senseAirCO2PPM;
  interrupts();
  
  doc[F("CO2ppm")] = test;
  doc[F("TemperatureC")] = BME280Temperature;
  doc[F("PressureHPa")] = BME280Pressure / 100.0F;
  //doc["AltitudeM"] = BME280Altitude;
  doc[F("HumidityPercent")] = BME280Humidity;

  return doc;
}
```

and you know it started working...

Honestly I have very big doubts about that problem is fixed. I'm not sure that it's just luck and that after several Loading scetch it will be broken again...
For me it looks like some problem with allocating value in memory. Maybe now it works just because fortune. I don't know. I want to try to remove `volatile` and `noInterrupts() - interrupts()` evening and see if it still works or not. But I'm afraid that if it depends on some memory stuff then it may work even after removing these lines...

Btw, 3 days ago when I was testing this first times there were not only 10000 but sometimes other numbers so maybe your guess about dangling pointer is right direction. I'm poor in C and memory management so need to investigate how to struggle with dangling pointer.

closed 2020-02-12 by bblanchon