bblanchon/ArduinoJson #1174: Can't use StaticJsonBuffer, must use DynamicJsonBuffer (V5.13.5).

effelle, 2020-01-27:
I have a simple code:

```
const size_t capacity = 1024;
DynamicJsonBuffer jsonBuffer(capacity);

const char* json = "{\"Temperature\":[\"A\",\"B\"],\"Humidity\":[\"C\",\"D\"]}";
JsonObject& root = jsonBuffer.parseObject(json);
const char* p1 = root[subsensortype][0];
const char* p2 = root[subsensortype][1];    
AddLog_P2(LOG_LEVEL_INFO, PSTR("JVAL: '%s', '%s', '%s'"),subsensortype, p1, p2);

```
where `subsensortype` is the input from another part of code.
I would like to use `StaticJsonBuffer<1024> jsonBuffer` as I'm using on other part of the code but sadly when I change it the program crash and enter an infinite loop.
Also, what would be the best "safeguard" if the `subsensortype` (as a key) will be not on json array ? 
For example with
```
 subsensortype = "Temperature"
```
all is fine, but if I check for a non existing key the program crash instead of returning null.

bblanchon, 2020-01-28:
Hi @effelle,

Please see [Why does my device crash or reboot?](https://arduinojson.org/v5/faq/why-does-my-device-crash-or-reboot/).
Let me know if you have any more questions.

Best regards,
Benoit

effelle, 2020-01-28:
Great thanks!

closed 2020-01-28 by effelle