bblanchon/ArduinoJson #1175: Updating JsonConfigFile

elC0mpa, 2020-01-29:
Hello, i am working on a project in which i need to update a configuration file which is loaded each time the program starts. In order to do this, I did the same as you in the example JsonConfigFile, but i have a question : each time you want to modify the file you delete it and generate it again, there is not a way to make this better , for example, without removing the file and Just reading it and later Updating it? I think it would increase the performance. Thanks for your time and for this great library

bblanchon, 2020-01-30:
Hi @elC0mpa,

Sure, you can do that.
The examples are just starting points, don't hesitate to modify it.
The take-away from JsonConfigFile is the `Config` struct: [it's better to use a custom data structure than to keep the `JsonDocument` in memory](https://arduinojson.org/v6/faq/why-must-i-create-a-separate-config-object/). 

Best regards,
Benoit

elC0mpa, 2020-01-30:
Hi @bblanchon, I knew this will be your answer, the thing is that i am using an esp32 and I am storing the file in its SPIFFS. When i open a file, it doesn't let me to open it in a read write Mode, because of this I am able to modify the json without creating it from scratch again and again but, the file itself is not updated. Any recommendation to give me? Maybe some useful link. Thanks in avance

bblanchon, 2020-01-31:
Hi @elC0mpa,

In the case study "Configuration in SPIFFS" in the last chapter of [Mastering ArduinoJson](https://arduinojson.org/book/), I show how to load/save a complex configuration object to a SPIFFS file.
Basically, you must open the file for reading, then close it, and reopen it for writing.

Best regards,
Benoit

elC0mpa, 2020-02-01:
Thanks @bblanchon, I really appreciate this comment, I Will let you know when I check it

closed 2020-02-12 by bblanchon