bblanchon/ArduinoJson #1176: Invalid Input

jorgher, 2020-02-01:


jorgher, 2020-02-01:
Hi Benoit
I'm sorry if I don't undestand the explantion to do the transmition between two arduino in series communication.

I am using arduino nano to collect data of temperature from a DHT11 sensor, also light intensitiy an humidity from soil,  once I have the values a DynamicJsonDocument doc(1024) is used to communiate with a ESP01 and sent the info to the web server an HTTP page.
here is the extract from the comm code:
```void loop() {
  lightValue = analogRead(LIGHTPIN);
  humidityValue = analogRead(SOILPIN);
  
  // Reading temperature or humidity takes about 250 milliseconds!
  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();
  // Read temperature as Fahrenheit (isFahrenheit = true)
  float f = dht.readTemperature(true);
  
  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) {
    Serial.println(F("Failed to read from DHT sensor!"));
    return;
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);
  
  Serial.println("  ");
  Serial.println(location);
  Serial.println(sensor);
  Serial.print(F("Humidity: "));
  Serial.print(h);
  Serial.print(F("%  Temperature: "));
  Serial.print(t);
  Serial.print(F("°C "));
  Serial.print(f);
  Serial.print(F("°F  Heat index: "));
  Serial.print(hic);
  Serial.print(F("°C "));
  Serial.print(hif);
  Serial.println(F("°F"));
  Serial.println("   ");
  Serial.print("Light reading = ");
  Serial.print(lightValue);     
  Serial.print(" lux ");
  Serial.println("  ");
  Serial.print("Humidity value = ");
  Serial.print(humidityValue);
  
  if (lightValue <= 1023 & lightValue >= 1000) {
    Serial.println("");
    Serial.print("Buenos Noches");
    digitalWrite(lightLED,HIGH);
    Serial.println("");
  }
  else {
    Serial.println("");
    Serial.print("Buenas Dias");
    digitalWrite(lightLED,LOW);
    Serial.println("");
  }

  if (humidityValue <= 1023 & humidityValue >= 500) {
    Serial.println("");
    Serial.print("No hay necesidad de regar");
    digitalWrite(RELAYpin,LOW);
    Serial.println("");
  }
  else {
    Serial.println("");
    Serial.print("Hay que regar la planta");
    regar();
    Serial.println("");
  }
  
  // Monitor serial communication
  while(Serial.available()) {
    message = Serial.readString();
    messageReady = true;
  }
  // Only process message if there's one
  if(messageReady) {
    // The only messages we'll parse will be formatted in JSON
    DynamicJsonDocument doc(1024); // ArduinoJson version 6+
    // Attempt to deserialize the message
    DeserializationError error = deserializeJson(doc,message);
    if(error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      messageReady = false;
      return;
    }
    
    if(doc["type"] == "request") {
      doc["type"] = "response";
      doc["humedad"] = dht.readHumidity();
      doc["tempC"] = dht.readTemperature();
      doc["tempF"] = dht.readTemperature(true);
      doc["HeatInx"] = dht.computeHeatIndex(f, h);
      doc["HeatInxF"] = dht.computeHeatIndex(t, h, false);
      doc["lightRead"] = analogRead(LIGHTPIN);
      doc["humedadPlanta"] = analogRead(SOILPIN);
           
      serializeJson(doc,Serial);
    }
    messageReady = false;
  }
  delay(6000);
}
```

Here is the arduino code:
```
#include <ArduinoJson.h>

double h=0, t=0, f=0, hic=0, hif=0, lightV=0, humidV=0;

void setup()
{
  Serial.begin(9600);
  // Send a JSON-formatted request with key "type" and value "request"
  // then parse the JSON-formatted response with keys "gas" and "distance"
}

void loop() 
{
  StaticJsonDocument<256> doc;
  // Sending the request
  doc["type"] = "request";
  serializeJson(doc,Serial);
  // Reading the response
  boolean messageReady = false;
  String message = "";
  while(messageReady == false) { // blocking but that's ok
    if(Serial.available()) {
      message = Serial.readString();
      messageReady = true;
    }
  }
  // Attempt to deserialize the JSON-formatted message
  DeserializationError error = deserializeJson(doc,message);
  if(error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }

  h = doc["humedad"];
  t = doc["tempC"];
  f = doc["tempF"];
  hic = doc["HeatInx"];
  hif = doc["HeatInxF"];
  lightV = doc["lightRead"];
  humidV = doc["humedadPlanta"];
  

  Serial.println(h);
  Serial.println(f);
  Serial.println(hic);
  Serial.println(hif);
  Serial.println(lightV);
  Serial.println(humidV);
}
```
I check the output in serial terminal from nano and everything look right, eve if I do {"type":"request"} I get the info back.
[image]

Now I connect to arduino uno using Rx and Tx pins inverted and just get.... Invalid Input on ESP serial console.

[image]

bblanchon, 2020-02-02:
Hi @jorgher,

I tried to find the issue, but there is way to much code up there.
If this problem is related to ArduinoJson, you should be able to remove everything else.
Please post the shortest possible program that demonstrates the bug.
Thank you for your understanding.

Best regards,
Benoit

jorgher, 2020-02-03:
The problem is with the communication between the arduinos, **Nano and Uno**, once I have the data coming from the sensor (In the Nano) I pass the values to a DinamicJsonDocument with the following sketch:

```
// Monitor serial communication
  while(Serial.available()) {
    message = Serial.readString();
    messageReady = true;
  }
  // Only process message if there's one
  if(messageReady) {
    // The only messages we'll parse will be formatted in JSON
    DynamicJsonDocument doc(1024); // ArduinoJson version 6+
    // Attempt to deserialize the message
    DeserializationError error = deserializeJson(doc,message);
    if(error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      messageReady = false;
      return;
    }
    
    if(doc["type"] == "request") {
      doc["type"] = "response";
      doc["humedad"] = dht.readHumidity();
      doc["tempC"] = dht.readTemperature();
      doc["tempF"] = dht.readTemperature(true);
      doc["HeatInx"] = dht.computeHeatIndex(f, h);
      doc["HeatInxF"] = dht.computeHeatIndex(t, h, false);
      doc["lightRead"] = analogRead(LIGHTPIN);
      doc["humedadPlanta"] = analogRead(SOILPIN);
           
      serializeJson(doc,Serial);
    }
    messageReady = false;
  }
```
To receive data in the arduino Uno I have a similar code showed in the previous message (arduino code) and the images attached are the results.

As you can see when there is a {"type":"request"} from the Arduino Uno the terminal say **deserialization() failed: InvalidInput**..

jorgher, 2020-02-09:
I already did that but still no response, my example is similar to the one
you posted on your video.

bblanchon, 2020-02-12:
I'm sorry but it's very hard to tell what is wrong with these snippets.
Maybe the transmission is buggy. Did you try to print the content of `message`?

closed 2020-02-25 by bblanchon