bblanchon/ArduinoJson #1177: Could it be local include ("x") instead of (<x>)?

rrakso, 2020-02-02:
Why here is this
https://github.com/bblanchon/ArduinoJson/blob/42b0d6a83d493bfa8b5a68e1a574fc60b5673036/src/ArduinoJson/Array/ArrayRef.hpp#L7
instead of somethnig like this ` #include "ArrayFunctions.hpp"`?

These files are in the same directory
I've changed these all occurences - all files are on my local computer, I can also make pull request with my changed files is so :wink:

rrakso, 2020-02-02:
[image]

bblanchon, 2020-02-02:
Hi @rrakso,

I switched to this form because of issue #1072.

If that's an issue for you, why don't you use the single header distribution?
You can find it in the [release page](https://github.com/bblanchon/ArduinoJson/releases/latest).

Best regards,
Benoit

closed 2020-02-12 by bblanchon

rrakso, 2020-02-13:
Hi @bblanchon what means _single header distribution_?
Can you help me understanding it?

(P.s. my problem is resolved, I just thought that probably I can help others)

bblanchon, 2020-02-13:
Single header means that the library fit in just one header: `ArduinoJson.h`
It's simpler to download this single file than to clone the repository.
See: https://arduinojson.org/v6/doc/installation/