bblanchon/ArduinoJson #1178: ARDUINOJSON_USE_DOUBLE == 1 not working as 64-bit precision.

kebaranas, 2020-02-03:
Arduino Model: Arduino Mega 2560

I am currently transmitting a json string using bluetooth low energy. I tried the example from the documentation however it only follows the 32-bit precision despite adding the #define ARDUINOJSON_USE_DOUBLE 1. Here is my code. `#define ARDUINOJSON_USE_DOUBLE 1
#include <ArduinoJson.h>
#include <TinyGPS++.h>

#define COIN_SLOT_PIN 2 

TinyGPSPlus GpsEncoder;

const uint8_t CAPACITY = 200;
const uint32_t PULSE_TIMEOUT = 500000;

volatile uint8_t coinSignal = 0;
uint16_t coinTotal = 0;

void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);
  Serial2.begin(9600);
  
  pinMode(COIN_SLOT_PIN, INPUT_PULLUP);

  attachInterrupt(digitalPinToInterrupt(COIN_SLOT_PIN), coinInserted, FALLING);
}

void loop() {
  while(Serial1.available() > 0) {
    GpsEncoder.encode(Serial1.read());
  }

  if (GpsEncoder.location.isUpdated()) {
    
    StaticJsonDocument<CAPACITY> doc; 
    doc["pi"] = 3.14159265359;
    serializeJson(doc, Serial2);
  }`

kebaranas, 2020-02-03:
This only replied a {"pi":3.141593}.

bblanchon, 2020-02-04:
Hi @kebaranas,

I don't have the hardware to check, but I'm pretty sure `sizeof(double) == 4` on Arduino Mega.
In other words, it's a hardware (or compiler, I'm not sure) limitation.

Best regards,
Benoit

kebaranas, 2020-02-04:
Hello @bblanchon,

You are right Arduino Mega does not support 8 bytes number. Is there some work around for this. I tried converting the double into char[] using dtostrf however when I serialize the output it only returns null. Meanwhile converting it into String only works for a while then later on returns a null. I also used this code. It works fine at first but then returns a null afterwards.

`
#include <ArduinoJson.h>
#include <TinyGPS++.h>

TinyGPSPlus GpsEncoder;

const uint8_t CAPACITY = JSON_OBJECT_SIZE(1);

StaticJsonDocument<200> doc; 

void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);
  Serial2.begin(9600);
}

void loop() {
  while(Serial1.available() > 0) {
    GpsEncoder.encode(Serial1.read());
  }

  if (GpsEncoder.location.isUpdated()) {
    char lngChar[9];
    String(GpsEncoder.location.lat(), 6).toCharArray(lngChar, 11); 
    Serial.println(lngChar);
    doc["lng"] = lngChar;    
    serializeJson(doc, Serial2);
  }
}
`

This would return {"lng":"14.165935"} then {"lng":null}

kebaranas, 2020-02-04:
Ok I already solved it. Thanks anyway.

closed 2020-02-04 by kebaranas