bblanchon/ArduinoJson #1179: Decoding Deeply Nested JSON (SignalK)

andyrbarrow, 2020-02-03:
I'm hoping for some tips - I wouldn't normally ask except I've been working on this on and off for weeks and I'm still struggling.
I'm decoding an incoming WebSockets stream of SignalK data for a marine navigation system. The updates are not fully known, as they come from different devices. When I tested with a doc.nesting() call it sometimes returned 5 and sometimes 6 nests.
Here's an example of a couple of inputs:
`{ 
   "context":"vessels.urn:mrn:signalk:uuid:eef4388d-ded6-4ed5-98f1-d7ddfa11ec83",
   "updates":[ 
      {
         "source":{ 
            "sentence":"MTW",
            "talker":"II",
            "type":"NMEA0183",
            "label":"nmeaTest"
         },
         "$source":"nmeaTest.II",
         "timestamp":"2020-02-03T21:05:28.354Z",
         "values":[ 
            { 
               "path":"environment.water.temperature",
               "value":280.15
            }
         ]
      }
   ]
}
{"context":"vessels.urn:mrn:signalk:uuid:eef4388d-ded6-4ed5-98f1-d7ddfa11ec83","updates":[{"source":{"sentence":"RMC","talker":"GP","type":"NMEA0183","label":"usbpuck"},"$source":"usbpuck.GP","timestamp":"2020-02-03T22:37:11.000Z","values":[{"path":"navigation.position","value":{"longitude":-105.29902716666666,"latitude":20.705448666666666}}]}]}
{ 
   "context":"vessels.urn:mrn:signalk:uuid:eef4388d-ded6-4ed5-98f1-d7ddfa11ec83",
   "updates":[ 
      { 
         "source":{ 
            "sentence":"RMC",
            "talker":"GP",
            "type":"NMEA0183",
            "label":"usbpuck"
         },
         "$source":"usbpuck.GP",
         "timestamp":"2020-02-03T22:37:11.000Z",
         "values":[ 
            { 
               "path":"navigation.position",
               "value":{ 
                  "longitude":-105.29902716666666,
                  "latitude":20.705448666666666
               }
            }
         ]
      }
   ]
}`
As you can see, the JSON is a mixture of objects and arrays. I'm struggling with understanding how to extract the data that I need. 
In this code, the "context" key is returned properly, but when I try to get the result of [updates][values][path] nothing is returned. I'm sure this is because the result is an array, but I don't know how to return it. Can you give me a pointer to an example or a call?
`    DeserializationError err = deserializeJson(doc, data);
    // Parse succeeded?
    if (err) {
      Serial.print(F("deserializeJson() returned "));
      Serial.println(err.c_str());
      return;
    }
    int doc_nests = doc.nesting();
    Serial.print("Nests: ");
    Serial.println(doc_nests);
    delay(500);
    JsonObject obj = doc.as<JsonObject>();
    const char* context = obj["context"];
    if (context != nullptr) {
      Serial.print("context: ");
      Serial.println(context);
    }
    const char* updates_path = obj["updates"]["values"]["path"];
    if (updates_path != nullptr) {
      Serial.print("updates_path: ");
      Serial.println(updates_path);
    }`

ewaldc, 2020-02-04:
@andyrbarrow, I have an application that parses (very large) JSON documents of unknown size and nesting depth (basically these are object trees of IOT devices).  A simplified version of this application is [**here**](https://github.com/ewaldc/OpenHAB-ESP).  In essence, I use a recursive function which descents all Json objects and arrays in search for certain objects which are used to populate a class instance that represents the state of all these IOT devices (since you can't maintain state in ArduinoJson itself).  In simplified form (assumes the top level is a Json Object and not an Array), the code looks like this:

```
void parseTree(const JsonObject obj, const char *pageId, Sitemap *sitemap) {
	static Page *lastPage = nullptr;
	for (const JsonPair pair : obj) {
		const char *key = pair.key().c_str();
		if (!strcmp_P(key, PSTR("linkedPage")) || !strcmp_P(key, PSTR("homepage"))) { // Page
			JsonObject pageObj = pair.value().as<JsonObject>();
			pageId = pageObj[F("id")];
			DbgPrintln(F("page: "), pageId);
			Page *page = new Page {pageId, (JsonVariant) pageObj, nullptr};
			if (lastPage) lastPage->next = page;
			else { // Homepage, set sitemap name
				sitemap->pageList = page;
				sitemap->name = pageId;
			}
			lastPage = page;
		} else if (!strcmp_P(key, PSTR("item"))) // Add item, to linked list
                        sitemap->itemList = new Item{pair.value().as<JsonObject>(), obj, pageId, sitemap->itemList};

		parseTree(pair.value().as<JsonObject>(), pageId, sitemap); //, pair.key);
		for (auto arr : pair.value().as<JsonArray>()) parseTree(arr.as<JsonObject>(), pageId, sitemap); //, pair.key);
	}
}
```

In this example, I am looking for _items_ and _pages_ (URI links named "linkedPage" or "homePage") anywhere in the Json object and building up linked lists of _pages_ and _items_ within a _sitemap_ class object as I encounter them.
If you need to know the level of nesting, you can use pass a _nesting_ variable which you increase every time you recursively descent down.   Don't forget to increase the nesting limit when parsing your Json strings (e.g. in my case up to 20 levels deep) :
`	DeserializationError err = deserializeJson(doc, f, DeserializationOption::NestingLimit(20));`

Hope this helps,
Ewald

andyrbarrow, 2020-02-04:
Thanks Ewald, I'll have to spend some time looking at this. My learning curve is very steep!

ewaldc, 2020-02-04:
No worries, we all have lots to learn...
The JSON you posted is not valid, so I could not post a working code example.  Below is a version that should print out the "content" and matching "path" strings if your Json string is an array of objects containing "content" keys.  If you pass a struct pointer as argument, you can build up a linked list of all "content" strings and their matching "path" values.

```
void parseTree(const const JsonVariant prototype) {
	if (prototype.is<JsonObject>()) {
	    const JsonObject& obj = prototype;
	    for (const JsonPair pair : obj) {
				const char *key = pair.key().c_str();
				if (!strcmp_P(key, PSTR("content")) { // We have a key named "content"
          Serial.print("context: ");
          Serial.println(pair.value().as<const char*>()); // value of "content"
        } else if (!strcmp_P(key, PSTR("path")) { // We have a key named "path"
          Serial.print("path: ");
          Serial.println(pair.value().as<const char*>()); // value of "path"
        }
				parseTree(pair.value().as<JsonObject>();
				for (auto arr : pair.value().as<JsonArray>()) parseTree(arr.as<JsonObject>());
			}
	} else if (prototype.is<JsonArray>()) {
		const JsonArray& arr = prototype;
		for (const auto& elem : arr) parseTree(elem.as<JsonVariant>(), sitemap); 
	}
}
```

andyrbarrow, 2020-02-04:
Thanks! I'm playing around with your code now.

In the mean time, this should be a valid JSON output from the instrument server:
`{
   "context":"vessels.urn:mrn:signalk:uuid:eef4388d-ded6-4ed5-98f1-d7ddfa11ec83",
   "updates":[
      {
         "source":{
            "sentence":"GLL",
            "talker":"GP",
            "type":"NMEA0183",
            "label":"nmeaTest"
         },
         "$source":"nmeaTest.GP",
         "timestamp":"2020-02-04T19:12:50.000Z",
         "values":[
            {
               "path":"navigation.position",
               "value":{
                  "longitude":138.467,
                  "latitude":-35.17916666666667
               }
            }
         ]
      }
   ]
}`

(not understanding why insert code in the message isn't showing this stuff properly formatted)

ewaldc, 2020-02-06:
OK, you could get away with simpler and faster code if it's just an array of such objects, since you don't have to deal with nested "context" objects.  Still, it will work.   Would need to see the most complex Json stream to get an idea of how much it can be simplified but
- it might be possible to take the _for_ loop dealing with the array of "context" objects out of the recursive function
- since you have only 2 objects at the toplevel ("context" and "updates"), you can simplify the recursion
- if "path" is always positioned at updates[0].values[0] then recursion is not needed

bblanchon, 2020-02-07:
Hi @andyrbarrow,

From your first message, it seems that you're ignoring the arrays.
This is the right way to access update path:

```c++
const char* updates_path = obj["updates"][0]["values"][0]["path"];
```

Also, remember that the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) is your friend.

Best regards,
Benoit

andyrbarrow, 2020-02-07:
Thanks very much Benoit!

andyrbarrow, 2020-02-07:
Thanks to both of you for your patience with me. Of course, it works perfectly. Onward!

bblanchon, 2020-02-08:
You're welcome Andy. Thank you for using ArduinoJson.
Please [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) on GitHub to show your support.

closed 2020-02-08 by bblanchon