bblanchon/ArduinoJson #1180: Serial Inputed Data not showing or Null when run

Thalinda, 2020-02-04:
Thank you very much for building this library and it's awesome. I am trying to build an automated message system using Arduino and nodeJS I am still Arduino part now I have tried manually passing my JSON data and it's working perfectly without any errors but when I try to pass my JSON file to Serial INPUT in Arduino it's responding with null or empty now this has worked perfectly before only when trying with Serial input it's not working I have searched every where and still can't find any thing now this my last hope really appreciate if you can help

```
`#include <ArduinoJson.h>
#include <SoftwareSerial.h>
SoftwareSerial mymessagecenter(7,8);
void setup() {
  mymessagecenter.begin(9600);
  Serial.begin(9600);  
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(9, OUTPUT);
  Serial.println("S-To Send Message");

  

}

void loop() {

  int     size_ = 0;
  String  payload;
  while ( !Serial.available()  ){}
  if (Serial.available() ){
    delay(50);
    payload = Serial.readStringUntil( '\n' );
    
    JsonMake(payload);
 
}

}

void JsonMake(String payload){
  Serial.println(payload);
  Serial.println("Testing");
  DynamicJsonDocument doc(1024);
  deserializeJson(doc, payload);
  JsonObject doc2 = doc.as<JsonObject>();
       delay(20000);
       const  String state = doc2["state"];
       Serial.println(state);
       const  String phon = doc2["number"];
       const  String fname = doc2["fname"];
       const  String lastname = doc2["lname"];
       const  String number = doc2["student_no"];
       const  String amount = doc2["amount"];
       const  String class_name = doc2["class_name"];
       
       digitalWrite(9, HIGH);
       Serial.println("Let's home this work");
       Serial.println(state);
       digitalWrite(9, LOW);
       if(state=="ture"){
           SendMessage(fname,class_name,phon,amount);
        }
  }
void SendMessage(String fname,String class_name,String phon,String amount){
    digitalWrite(9, HIGH); 
    mymessagecenter.println("AT+CMGF=1");
    delay(1000);
    mymessagecenter.println("AT + CMGS = \"" + phon +"\"\r");
    delay(1000);
    mymessagecenter.println("Hello,"+fname+" Today 7 student(s) paid class fee and "+amount+"/= has recived as your "+class_name+" class payment please collect it from front office ** ");
    delay(2000);
    mymessagecenter.println((char)26);
    digitalWrite(9, LOW);
  }

////{state:"true",number:"xxxxxxxxx",fname:"xxxxxxxx",student_no:"7",amount:"7500",class_name:"Sinhala"}` JSON That i am passing
```
```

> 22:49:08.580 -> Testing
> 22:49:28.561 -> null
> 22:49:28.561 -> Let's home this work
> 22:49:28.561 -> null

bblanchon, 2020-02-05:
Hi @Thalinda,

Please check the value returned by [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).

Also, you don't need to use a temporary `String` because [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) is able to pull the data directly from `SoftwareSerial`.

Best regards,
Benoit

Thalinda, 2020-02-05:
Thank you so much for quick reply now it's working perfectly now as I can the issue is assing Serial input to string and passing that to Deserialization when I passing it directly without Variables it's working fine now there is another issue now it's sending message and everything ok but after sending message it's shows the message 

> 17:51:01.213 -> deserializeJson() failed: IncompleteInput
 
**### This my new code realy appriciate it if you can give me any suggetions 👍**
```
#include <ArduinoJson.h>
#include <SoftwareSerial.h>
SoftwareSerial mymessagecenter(7,8);
 

void setup() {
  mymessagecenter.begin(9600);
  Serial.begin(9600);  
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(9, OUTPUT);
  Serial.println("S-To Send Message");

  

}

void loop() {

  int     size_ = 0;
  String  payload;
  while ( !Serial.available()  ){}
  if (Serial.available() ){
   while (!Serial.available())
    delay(50);
    StaticJsonDocument<512> doc;

    DeserializationError error = deserializeJson(doc, Serial);
    delay(1000);
    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      return;
    }
    const char* state = doc["state"];
    String price = doc["amount"];
    String fname = doc["fname"];
    String phone = doc["number"];
    String number_of_student = doc["student_no"];
    String class_name = doc["class_name"];
    SendMessage(fname,number_of_student,class_name,phone,price);
  
}

}

void SendMessage(String fname,String count,String class_name,String phon,String amount){
  Serial.println(phon);
    digitalWrite(9, HIGH); 
    mymessagecenter.println("AT+CMGF=1");
    delay(1000);
    mymessagecenter.println("AT + CMGS = \"" + phon +"\"\r");
    delay(1000);
    mymessagecenter.println("Hello,"+fname+" today "+count+" student(s) paid class fee and "+amount+"/= rupees recived as your "+class_name+" class payment please collect it from front office ** ");
    delay(2000);
    mymessagecenter.println((char)26);
    digitalWrite(9, LOW);
  }
```

bblanchon, 2020-02-06:
Hi @Thalinda,

It could be a timeout, please read: [How to change the timeout when reading from a Stream?](https://arduinojson.org/v6/how-to/change-the-timeout/)

Best regards,
Benoit

closed 2020-02-25 by bblanchon