bblanchon/ArduinoJson #1181: Intellisense Errors in VS2019

avidNewb, 2020-02-08:
I'm using VS2019 with vMicro installed to compile code for an MH-ET Live ESP32 MiniKit.  I'm seeing an issue that was mentioned on a closed issue before.  As soon as I add ArduinoJson I receive hundreds of intellisense errors.  I have not been able to make these errors go away.  The code does compile but this renders intellisense completely useless.

I've also noticed that intellisense doesn't recognize classes and functions from ArduinoJson.  For instance DynamicJsonDocument does not show up in the list when I begin typing and appears with a red line underneath after typing.  I can put ARDUINOJSON_NAMESPACE:: in front of it and it shows up.  This doesn't resolve the functions though such as serializeJson because they are not in a namespace.

avidNewb, 2020-02-08:
After checking again, I noticed that the functions are in a namespace so I don't know why it doesn't fix them.

bblanchon, 2020-02-08:
Hi @avidNewb,

I installed Visual Micro on VS2019, and Intellisense seems to be working fine.

> I'm seeing an issue that was mentioned on a closed issue before

Which issue?

>  I receive hundreds of intellisense errors.

What errors?

All functions and classes of ArduinoJson are in the namespace `ArduinoJson`.
When you include `ArduinoJson.h` (and not `ArduinoJson.hpp`), a `using namespace ArduinoJson;` brings all symbols in the global namespace.

`ARDUINOJSON_NAMESPACE` is a macro that dynamically defines the name of another private namespace. The generated name includes the version number and the compile-time options, which allows embedding multiple versions of the library in the same binary. The actual name is something like `ArduinoJson6141_0000010`. Unfortunately, Intellisense shows this namespace instead of the public one, and I don't think there is any way to fix this.

Best regards,
Benoit

avidNewb, 2020-02-08:
I was referring to https://github.com/bblanchon/ArduinoJson/issues/483

Interesting. I have included ArduinoJson.h.  Putting "using namespace ArduinoJson;" shows an error but it does make most of the other errors go away in my project.  Using your progmem example though it doesn't fix the errors. I get 311 errors.  If I add "using namespace ArduinoJson;" it is then 312.  Including, "ARDUINOJSON_NAMESPACE" makes the classes show up but not the functions, even in the example project and does nothing for the other errors.  I'm going to move to another machine just to confirm it's not an issue with this one.

avidNewb, 2020-02-08:
I apologize.  My other machine doesn't have this issue at all.  Interesting that it's only with this library.  Updating Arduino IDE and reinstalling vMicro hasn't helped.  I'm currently repairing my IDE.  If I figure out what is causing it, I'll comment here in case anyone else has the issue.

bblanchon, 2020-02-09:
> I'll comment here in case anyone else has the issue.

Excellent idea! 
Let me know I can help you.

Tintin4000, 2020-02-21:
I am having exactly the same issue.  I upgrade from ArduinoJson 6.13 (no issue with intellisense at that time) to 6.14.1 (which exhibit the issue).  I will try to downgrade to 6.13.

Tintin4000, 2020-02-21:
I can confirm that the downgrade to 6.13 fixed the intellisense issue in VS2019.  I have tried with 6.14 and 6.14.1, identical behavior.

avidNewb, 2020-02-21:
> I can confirm that the downgrade to 6.13 fixed the intellisense issue in VS2019. I have tried with 6.14 and 6.14.1, identical behavior.

Can you confirm what board you are using?  I think we have narrowed it down to ESP32 boards in the vmicro forum.

https://www.visualmicro.com/forums/YaBB.pl?num=1581276046/20#23

bblanchon, 2020-02-21:
Hi,

@Tintin4000, thank you for narrowing down the issue to version 6.14.0 and ESP32.

ArduinoJson 6.14.0 automatically enables support for `std::string` and `std::stream` when it detects the required headers.
It seems that enabling `std::stream` triggers a bunch of errors.

I don't know if it's a problem with ArduinoJson or with the ESP32 core.

Until we get a better understanding of the problem, you can manually disable the support for `std::stream` by setting [ARDUINOJSON_ENABLE_STD_STREAM](https://arduinojson.org/v6/api/config/enable_std_stream/) to `0`.

Best regards,
Benoit

Tintin4000, 2020-02-21:
Yes, indeed I am using ESP32.  The previous post fix the intellisense issue.  Since I am not using stream in my code, I am fine.  Thank you so much for your help and a great library.  I revert back to 6.14.1.

smuellener, 2020-03-25:
We also experience similar intellisense issues using VisualStudio, VisualMicro, ESP32 and including ArduinoJson.h (all in very recent versions). Intellisense is working for most cases but not all e.g.

- "DynamicJsonDocument doc(256)" is recognized

- "serializeJson(doc, buffer, sizeof(buffer))" is not recognized

Is this the same issue described in this topic?

bblanchon, 2020-03-25:
@smuellener, yes, that's the same setup.
Did you try to set [ARDUINOJSON_ENABLE_STD_STREAM](https://arduinojson.org/v6/api/config/enable_std_stream/) to `0`?

smuellener, 2020-03-25:
@bblanchon yes I have in the following way:
 
#define ARDUINOJSON_ENABLE_STD_STREAM 0
#include <ArduinoJson.h>

instead of 

#include <ArduinoJson.h>

in the .Ino file. but with no success. We are including ArduinoJson.h in different other files (libraries) in the solution. I did not do this in all the files (I guess this is not needed).

I will try to investigate a bit more. If you would want to have a look yourself but do not have a setup, I also could arrange a way to remotely access.

Connect-and-Exchange, 2020-06-14:
Following - tested with 6.15.2 and still the problem persists. Downgraded to 13 and the intellisense works again

smuellener, 2020-07-15:
@bblanchon How can we improve this? We can not really make out where the issue is.. is it even a problem of this library or a problem of VisualStudio?

smuellener, 2020-07-17:
@bblanchon ok after trying out diferent things the following worked (setting both to 0 resulted in the problem of #1045 )

ARDUINOJSON_ENABLE_STD_STREAM=0;ARDUINOJSON_ENABLE_STD_STRING=1

bblanchon, 2020-07-22:
I think this is a problem with the ESP32 headers.
We can easily reproduce the problem without ArduinoJson, simply by including the [standard `<ios>` header](https://en.cppreference.com/w/cpp/header/ios):

```c++
#include <ios>
```

Also, I noticed that VisualMicro/Intellisense tends to pick the wrong headers when the Arduino core doesn't provide them.
For example, if you try to include `<ios>` on AVR, instead of saying `cannot find source file "ios"`, it falls back to the classic MSVC headers and therefore displays a bunch of Intellisense errors.

I was not able to find a workaround, so I don't think I can fix this error.
@smuellener, could you report this issue to the corresponding projects?

bblanchon, 2020-09-25:
Guys, is there any update on this?

smuellener, 2020-09-25:
> Guys, is there any update on this?

To be honest, I do not know where to raise this issue and what headers are responsible for this. But what I can say is that it works for us with the defines described above.

bblanchon, 2020-09-25:
Any chance an update might have resolved the issue without you noticing it?

MartinNohr, 2020-09-28:
I finally got around to doing some more testing on my system here. I have not found any combination of namespaces and defines as discussed above that work for me. I have however noticed that it is only failing when I compile for HelTec ESP32 WiFi Kit 32. They have their own .h file and the conflicts are around the String definition. As long as I don't use any of the json namespaces it compiles, but of course intellisense gives warnings. If I use the json namespace I get "String is ambiguous" and it won't compile/link.

bblanchon, 2020-09-28:
@MartinNohr, ArduinoJson contains a `String` class in the `ARDUINOJSON_NAMESPACE`.
This namespace is private and is reserved for internal use.
In your application, you should use only the `ArduinoJson` namespace.

MartinNohr, 2020-09-30:
I have this
#include <ArduinoJson.h>
using namespace ArduinoJson
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>
#include <ringbufcpp.h>

I get this error when compiling:
esp_bt_defs.h: 22:1: error: expected ';' before 'extern
   extern "C" {
  
If it remove the namespace it compiles just fine.
Here is part of that file:
#include <stdint.h>
#include <stdbool.h>

#ifdef __cplusplus
extern "C" {
#endif

#define ESP_BLUEDROID_STATUS_CHECK(status)           \
    if (esp_bluedroid_get_status() != (status)) {    \
        return ESP_ERR_INVALID_STATE;                \
    }

/* relate to BT_STATUS_xxx in bt_def.h */
/// Status Return Value
typedef enum {
    ESP_BT_STATUS_SUCCESS       = 0,            /* relate to BT_STATUS_SUCCESS in bt_def.h */
    ESP_BT_STATUS_FAIL,                         /* relate to BT_STATUS_FAIL in bt_def.h */

Anything else you'd like to see? I only see this error with the Heltec ESP32 WIFI kit.

bblanchon, 2020-09-30:
Hi @MartinNohr,

You're missing a semicolon after `using namespace ArduinoJson`, that's probably the reason for the error.
Anyway, you don't need to add this line, since `ArduinoJson.h` already imports the namespace.

BTW, are you using VisualMicro, and are you seeing Intellisense errors?
It looks like your problem is unrelated to the current issue.

Best regards,
Benoit

MartinNohr, 2020-10-01:
Well, that's embarrassing, a missing ; . I've been a programmer using C and C++ (and others) for 40 years. :-) I must be getting old...
Anyway, yes, I'm using VisualMicro.  Yes, I see intellisense errors in that symbols are declared as undefined, even though they compile just fine. Here's a picture showing some undefined symbols.
[image]
The red squiggly underlines means undefined symbol, but it compiles just fine. Very strange.

MartinNohr, 2020-10-01:
And even more curious, if I select one of those, F12 does take me to the definition correctly, which is what it is supposed to do.

MartinNohr, 2020-10-01:
This might be interesting.
[image]

vortex314, 2021-01-14:
Intellisense is basing its syntax support on the includes that are defined in .vscode/c_cpp_properties.json file. Adapting this file to point to the right include directories helps a lot. 
[image]

bblanchon, 2021-05-19:
Maybe it's time to close this issue... is the error still occurring?

Connect-and-Exchange, 2021-05-19:
Yes it is…

Martin Meuwese
***@***.***
 | 
m: +31642679706

CEO, Connect and Exchange
t: +31223222100 | f: +31223720717
e: ***@***.*** | w: www.cande.eu
a: Keizersgracht 65, 1781BA, Den helder

bblanchon, 2021-05-19:
I showed [in a previous comment](https://github.com/bblanchon/ArduinoJson/issues/1181#issuecomment-662277972) that this bug is not specific to ArduinoJson.
Did anyone report it to VisualMicro?

bblanchon, 2023-07-29:
In [ArduinoJson 6.21](https://arduinojson.org/news/2023/03/14/arduinojson-6-21-0/), I replaced `ARDUINOJSON_NAMESPACE` with an inline namespace, which may fix this issue.
Does this problem still occur with 6.21.3?

closed 2023-10-09 by bblanchon