bblanchon/ArduinoJson #1182: Possible issue with suggested invocation of DynamicJsonDocument.shrinkToFit()

memr3x, 2020-02-09:
This is not an issue with the library itself but the example invocation of using the new shrink to fit feature. In both the change-log post and documentation it's suggested that `ESP.getFreeHeap()` can be used to get the maximum allocation size on the ESP platform. This may result in a  failed allocation even with the 4Kb size reduction demonstrated as the `ESP.getFreeHeap()` method returns total free heap regardless of fragmentation.

Instead `ESP.getMaxFreeBlockSize()` should be used for **ESP8266** and `ESP.getMaxAllocHeap()` for **ESP32** (unfortunately the call differs between the two platforms, and obviously this would also vary between non-Espressif platforms.) These calls actually return the largest single block that can be allocated rather than the total accumulated free heap space.

bblanchon, 2020-02-09:
Hi @memr3x,

Thank you very much for this remark.
I forgot about this function, even though I used it to make the measurements for my article on [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html).

I updated the examples in arduinojson.org to use these functions.

Best regards,
Benoit

closed 2020-02-09 by bblanchon

ewaldc, 2020-02-09:
@bblanchon, @memr3x 
Apologies, it was actually me who posted a code fragment using `ESP.getFreeHeap()` and Benoit was so kind to develop the `shrinkToFit()` function for my use case.   I have updated the code since to use `ESP.getMaxFreeBlockSize()` after hitting a failed allocation, but forgot to report this back.  
My bad...