bblanchon/ArduinoJson #1183: Slow deserialisation

jan-j, 2020-02-10:
Hi,

First of all thanks for the great library!

We're trying to move most of our animation logic code from Arduino low level code into the high level side of things. That means that we need to make the communication fast enough to be able to control the animation quickly enough.

I've run into an issue with deserializeJson, which seems to be pretty slow at reading from Serial. When I try deserialisation of single JSON structure it takes 500-1000ms just for 
`DeserializationError err = deserializeJson(doc, Serial);`

When I try deserialisation of consecutive multiple JSON structures it takes 500-1000ms just for the first JSON and ~50ms for following ones (which still very slow).

I tried using `ReadBufferingStream` which helped speed things up when there was nothing in the buffer, but doesn't help with the deserialisation itself.

Is this the speed to expect? Or am I doing something wrong?

bblanchon, 2020-02-11:
Hi,

ArduinoJson stops reading as soon as the document ends (most of the time, it's when it sees a closing brace), so [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) should not wait for a timeout.

Maybe the timeout occurs because the serial buffer is empty? You should call `Serial.available()` to make sure there is something in there.

In the case study "Recursive Analyzer"  in [Mastering ArduinoJson](https://arduinojson.org/book/), I use two loops: one to flush the buffer (in case it contains garbage from a previous run) and a wait loop:

```c++
// Flush Serial input buffer
while (Serial.available())
  Serial.read();

// Waiting for a JSON document in Serial
while (!Serial.available())
  delay(50);

// Parsing
DynamicJsonDocument doc(1024);
DeserializationError error = deserializeJson(doc, Serial);
```

Best regards,
Benoit

jan-j, 2020-02-11:
Thanks, your answer was certainly helpful! I've realised that the thing throttling down the speed was the serial baud rate which I increased to 115200 and deserialize times went from 25-50ms to 4-5ms.

Messages that I'm sending in batches looks like this
```json
{"type":"setPixelColor","data":{"i":0,"color":[85,0,0]}}
```

I wonder if 4-5ms per message like this is normal. For our use case it would need to be 10 times faster than that realistically.

bblanchon, 2020-02-11:
According to [my measurements](https://arduinojson.org/2019/11/19/arduinojson-vs-arduino_json/),  ArduinoJson is pretty fast, especially for small payloads like that.

If I'm not mistaken, at 115200 bauds, the transmission takes 3.8 ms.

To get a 10-fold speed improvement, you must use a compact binary format.

jan-j, 2020-02-11:
Thanks, that's really helpful. Unfortunately that also means that we'll have to change the approach at least a little.

closed 2020-02-11 by jan-j