bblanchon/ArduinoJson #1184: Support for GCC 9 ?!

soeth16, 2020-02-13:
Hi at all, 
In order to compile the framework with GCC 9.2 I made following changes:

src/ArduinoJson/Object/MemberProxy.hpp
```C++
...
template <typename TObject, typename TStringRef>
class MemberProxy : public VariantOperators<MemberProxy<TObject, TStringRef> >,
                    public Visitable {
  typedef MemberProxy<TObject, TStringRef> this_type;

 public:
  MemberProxy(const MemberProxy&) = default; // GCC 9
...
```

src/ArduinoJson/Array/ElementProxy.hpp
```C++
...
template <typename TArray>
class ElementProxy : public VariantOperators<ElementProxy<TArray> >,
                     public Visitable {
  typedef ElementProxy<TArray> this_type;

 public:
  ElementProxy(const ElementProxy&) = default; // GCC 9
...
```

extras/tests/JsonDocument/BasicJsonDocument.cpp:
```C++
...
class SpyingAllocator {
 public:
  SpyingAllocator(const SpyingAllocator&) = default; // GCC 9
...
```

closed 2020-02-13 by bblanchon

bblanchon, 2020-02-13:
Hi @soeth16,

Thank you for reporting this.
I fixed this warning in the main branch.

Best regards,
Benoit

bblanchon, 2020-03-22:
This fix was included in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

commit d2a67f36 (2020-02-13): Fixed "deprecated-copy" warning on GCC 9 (fixes #1184)