bblanchon/ArduinoJson #1187: Values of json structure disappear after the first time.

nvecchio1, 2020-02-17:
Hi,
i'm using ArduinoJson V6 to programming my esp32 with Platform.io.
Here my code:
`const int capacity = JSON_ARRAY_SIZE(2) + 4 * JSON_OBJECT_SIZE(2);
StaticJsonDocument<capacity> doc;
// Add the "location" object
JsonObject location = doc.createNestedObject("location");
location["lat"] = 48.748010;
location["lon"] = 2.293491;
// Add the "feeds" array
JsonArray feeds = doc.createNestedArray("feeds");
JsonObject feed1 = feeds.createNestedObject();
feed1["key"] = "a1";
feed1["value"] = 2345;
JsonObject feed2 = feeds.createNestedObject();
feed2["key"] = "a2";
feed2["value"] = 2443;
Serial.println();
serializeJsonPretty(doc, Serial);
delay(5000);`
When i running the program, the values are good. But at the second time the structure of json is empty , there are only name of json object nested in doc.
[image]

nvecchio1, 2020-02-17:
And if i don't naming nested objects it come all good.

[image]

How can i fix it?

bblanchon, 2020-02-17:
Hi @nvecchio1,

Please read:

* [I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/).

Best regards,
Benoit

closed 2020-02-25 by bblanchon