bblanchon/ArduinoJson #1188: Parsing JSON

zedb123, 2020-02-18:
Thanks for the book - really useful for a newbie ..

defined : 

	char doneMsg[] = "{\"state\" : { \"reported\" : { \"power\" : \"off\" } } }";

***************
NOW I TRY TO PARSE BY DESERIALISATION
***************

	const int jsonSize = JSON_OBJECT_SIZE(3) + 100;
	DynamicJsonDocument data(jsonSize);
	  
	DeserializationError err = deserializeJson(data, doneMsg);
	if (err)
	{
		Serial.print(F("deserializeJson() failed: "));
		Serial.println(err.c_str());
	}
	JsonObject obj = data.as<JsonObject>();
  	const char* state_reported_power = obj["state"]["reported"]["power"]; // "on"
	const char* state = obj["state"]["reported"]; //  should be "reported"
    Serial.print("*");
    Serial.print(state_reported_power);
    Serial.println("*");

    Serial.print("*");
    Serial.println(state);
    Serial.println("*");

*************
I DONT GET A VALUE FOR  state ... however i do for state_reported_power 
*************

I know it's obvious , I just can't see it !

Thanks

bblanchon, 2020-02-19:
Hi @zedb123,

I'm not sure I understand your problem.
If you're trying to get the value "off", then you must write:

```c++
const char* state_reported_power = doc["state"]["reported"]["power"]; // "off"
```

If that's not what you want, please clarify your question.

Best regards,
Benoit

zedb123, 2020-02-19:
Hi Benoit,
That piece works .. As I understand it, protocol for AWS IOT is device sends "reported" back to AWS to confirm state.Client sends "Desired" to AWS to change state
The device (using the ArduinoJson) receives a JSON message when shadow updated to "reported". The device will obviously receive message when Client issues "desired"
I want to be able to see if the message is "reported" or "desired".
Does that make sense?
Thanks

bblanchon, 2020-02-19:
Hi @zedb123,

If I understand correctly, your goal is to differentiate between:

```json
{"state":{"reported":{"power":"off"}}}
```

and
 
```json
{"state":{"desired":{"power":"off"}}}
```

If that's what you want, you just need to test the presence of the "reported" or "desired" key:

```c++
if (doc["state"]["reported"]) {
  // ...
}

if (doc["state"]["desired"]) {
  // ...  
}
```

Alternatively, you could write:

```c++
const char* reported_power = doc["state"]["reported"]["power"];
const char* desired_power = doc["state"]["desired"]["power"];

if (desired_power) {
  // ...
}
```

Now, if you just want to change the state of your device according to the desired power, you can write:

```c++
JsonVariant desired_power = doc["state"]["desired"]["power"];

if (desired_power == "on") {
  // switch ON
}

if (desired_power == "off") {
  // switch OFF
}
```

Best regards,
Benoit

zedb123, 2020-02-19:
Thanks Benoit .. That's exactly what I wanted.

closed 2020-02-19 by zedb123

bblanchon, 2020-02-20:
You're welcome, @zedb123, thank you for using ArduinoJson.
Don't forget to [start the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support.
Good luck with your project.