bblanchon/ArduinoJson #1189: Problems on migrating for V.5 to V.6 (ESP8266 HVAC / modbus) program)

jascdk, 2020-02-19:
Hi Benoit 

I am currently struggling to get a program up and running with Arduino JSON v.6 .

Im a noob at coding, but hoping you / others could help me on the way.

I have a program that I would like to put on a esp8266 microcontroller - it is going to control our local HVAC over modbus.

I simply can't figure out how to convert it - I have looked everywhere and also you nice videos on YouTube. I don't have any coding background at all :/

Looking forward to hear from you:)

```
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ArduinoOTA.h>
#include <ModbusMaster.h>
#include <PubSubClient.h>
#include "configuration.h"
#if SERIAL == SERIAL_SOFTWARE
#include <SoftwareSerial.h>
#endif

#define SERIAL_SOFTWARE 1
#define SERIAL_HARDWARE 2

#define HOST "NilanGW-%s" // Change this to whatever you like. 
#define MAXREGSIZE 26
#define SENDINTERVAL 30000 // normally set to 180000 milliseconds = 3 minutes. Define as you like
#define VENTSET 1003
#define RUNSET 1001
#define MODESET 1002
#define TEMPSET 1004

#if SERIAL == SERIAL_SOFTWARE
SoftwareSerial SSerial(SERIAL_SOFTWARE_RX, SERIAL_SOFTWARE_TX); // RX, TX
#endif

const char* ssid = WIFISSID;
const char* password = WIFIPASSWORD;
char chipid[12];
const char* mqttserver = MQTTSERVER;
const char* mqttusername = MQTTUSERNAME;
const char* mqttpassword = MQTTPASSWORD;
WiFiServer server(80);
WiFiClient client;
PubSubClient mqttclient(client);
static long lastMsg = -SENDINTERVAL;
static int16_t rsbuffer[MAXREGSIZE];
ModbusMaster node;

String req[4]; //operation, group, address, value
enum reqtypes
{
  reqtemp = 0,
  reqalarm,
  reqtime,
  reqcontrol,
  reqspeed,
  reqairtemp,
  reqairflow,
  reqairheat,
  requser,
  requser2,
  reqinfo,
  reqinputairtemp,
  reqapp,
  reqoutput,
  reqdisplay1,
  reqdisplay2,
[... 421 lines cut ...]

          for (int i = 0; i < regsizes[r]; i++)
          {
              char *name = getName(r, i);

              if ((rsbuffer[i] & 0x00ff) == 0xDF) {
                text += (char)0x20; // replace degree sign with space
              } else {
                text += (char)(rsbuffer[i] & 0x00ff);
              }
              if ((rsbuffer[i] >> 8) == 0xDF) {
                text += (char)0x20; // replace degree sign with space
              } else {
                text += (char)(rsbuffer[i] >> 8);
              }
              mqname += (char *)name;
          }
          mqttclient.publish(mqname.c_str(), text.c_str());
        }
      }
      lastMsg = now;
    }
  }
}
```

bblanchon, 2020-02-20:
Hi @jascdk,

I'd sincerely like to help you out.
To do that, I need you to:

1. explain what the problem is,
2. reduce the length of the code (50 lines max).

Thank you for your understanding.

Best regards,
Benoit

jascdk, 2020-02-25:
Sure ☺️ I’m abroad for a few days and will do it when I get home again :) thanks a lot

jascdk, 2020-03-04:
hi @bblanchon :)

First of all thanks for all your great work with the library.

In my house we have a HVAC system that communicates via modbus. I have attached an Wemos D1 mini to it to control it.

It uses this maincode : https://pastebin.com/QnEvGme9 (so you can se it as an reference)

1: My main problem was that the code used json 5 and I would update it to version 6, so it uses the latest version of the library. Make less trouble:)

I have tried asking help from my wife (she codes at her work - but often won't help me, as she hates all my home automation stuff :D ) 

I did though manage to persuade her to look at my problem. Which gave this result 

[image]

It actually works, but I can't figure out why she did not change all the "root" to "doc" instead like mentioned in you videos.

She could not give me an answer of this.

Does it looks right to you?

bblanchon, 2020-03-04:
Hi @jascdk,

What she wrote seems correct.

I use the name "doc" for the complete document and "root" for the root object, so using both names is fine.

There is just one thing that surprises me though: it's that the `JsonDocument` is passed by value and not by reference. To me it should be:

```c++
JsonObject HandleRequest(JsonDocument& doc);
```

To me, passing by value is not appropriate here because it's not a real copy.
It accidentally works because `JsonDocument` copies its internal pointers.
I'm surprised this was even possible, I'll fix that in the next release.

Lastly, you don't need to return the root object from `HandleRequest()` because the `JsonDocument` contains all the required information. Change the signature of `writeResponse()` to:

```c++
void writeResponse(WiFiClient& client, const JsonDocument& doc);
```

Does that make sense?

Best regards,
Benoit

jascdk, 2020-03-04:
hi @bblanchon :)

Thanks for a very quick reply. I will try make the changes you describe. And then I will talk to my wife and hear her about it, and come back to her with her answer:)

Thanks a lot!

Cheers:)

closed 2020-03-20 by bblanchon

bblanchon, 2020-03-22:
Copy-contructor of `JsonDocument` is now hidden in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

jascdk, 2020-03-28:
@bblanchon Thanks Benoit:) I got my wife to help with the fixes as you mentioned - it just works now - thanks alot;)

All the best

Jacob / Denmark