bblanchon/ArduinoJson #1190: Append json object to file containing json array without reading everything into memory

jpmeijers, 2020-02-20:
I've seen two previous issues where a user wanted to append a json object to a file. The answers go on telling how to read the content of the file into memory, parse it as a json array, adding the new object to the array, and then writing out the complete array.

Is there a way in which one can append the new object to the end of the file, without having to read in the entire contents of the file? I'm imagining one can scan to the end of the file, see the last character is a `]`, remove the last character, append a ',' followed by the new json object, followed by a ']'.

Another way would be to add a '[' to the file when it is created, append the json object, followed by a ','. When you know you are done with this file and you will never append data again (date of the day), append a ']'.

Has anyone done this before, and how did you do this? The file I'm writing to is on an SD card (using the SdFat library), and one will likely have to keep in mind that reading and writing sd cards are done in 512 byte blocks.

A naive approach would be to write the json object to the file, followed by two newlines. In other words a newline separated list of json objects.

bblanchon, 2020-02-21:
Hi @jpmeijers,

It seems that you want to use [JSON Lines](http://jsonlines.org/) / [NDJson](http://ndjson.org/) format.
You can use ArduinoJson to do that with the techniques you describe.
I don't know if anyone already did that, but it looks feasible to me.

However, I think it would be simpler to create one file per entry.
This way you add entries without affecting the previous ones.
And it allows random access to any element without loading the complete array.

Best regards,
Benoit

jpmeijers, 2020-02-22:
Thanks for those two links. I did not know there was a standard for newline delimited json. I feel better using that approach now.

On my embedded platform I only want to write out logfiles to an sd card. These logfiles do not need to be read back by the microcontroller. If we need to look at the logfiles we will remove the sd card and copy the files to a computer for processing.

closed 2020-02-22 by jpmeijers

bblanchon, 2020-02-22:
You're welcome, @jpmeijers.
Good luck with your project.
Don't hesitate to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)