bblanchon/ArduinoJson #1191: JsonObject not maintaining char[] data in nested scope

kawinie, 2020-02-22:
```c++
StaticJsonDocument<200> doc;
JsonArray array = doc.to<JsonArray>();
{
    JsonObject obj = array.createNestedObject();
    char text[]	   = "Panda Cubs;";
    obj["text"].set(text);
}
serializeJsonPretty(array, Serial);
```
The above code incorrectly outputs 
```json
[
  {
    "text": "(sometimes garbage, other times empty)"
  }
]
```

Adding an explicit (char *) cast to "text" outputs a correct result.
```c++
StaticJsonDocument<200> doc;
JsonArray array = doc.to<JsonArray>();
{
    JsonObject obj = array.createNestedObject();
    char text[]	   = "Panda Cubs;";
    obj["text"].set((char *)text);
}
serializeJsonPretty(array, Serial);
```
```json
[
  {
    "text": "Panda Cubs"
  }
]
```
Seems like the problem is that JsonDocument thinks `char[]` is `const char *`, in which case the document simply just copy over the pointer.

bblanchon, 2020-02-22:
Hi @kawinie,

Thank you very much for reporting this bug.
It seems to be limited to `MemberProxy::set()`.

Indeed, if you use `operator=()`, it works as expected:

```c++
obj["text"] = text;
```

Best regards,
Benoit

closed 2020-02-22 by bblanchon

bblanchon, 2020-03-22:
This fix was included in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

commit af0edecd (2020-02-22): Fixed MemberProxy::set(char[]) not duplicating the string (fixes #1191)