bblanchon/ArduinoJson #1192: merge example from Bag of Tricks will not compile (iterator problem)

nsbawden, 2020-02-24:
The basic merge example from Bag of Tricks (and elsewhere on the web) will not compile. In fact, all code examples using iterators for ArduinoJson, even older C style, will not compile. Tried in both web based Arduino IDE and in Arduino Pro Alpha IDE with same error.

ArduinoJson library version 6.14.1

######### Basic Compile Error

/tmp/451904423/ArduinoJsonTest/ArduinoJsonTest.ino:5:13: error: no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject {aka ArduinoJson6141_0000010::ObjectRef}' and '<unresolved overloaded function type>')

######### Code

`#include <ArduinoJson.h>

void merge(JsonObject& dest, const JsonObject& src) {
    for (auto kvp : src) {
        dest[kvp.key] = kvp.value;
    }
}

void setup() {
    
}

void loop() {
    
}
`
######### Full Compile output

/home/builder/opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -hardware /home/builder/opt/arduino-builder/hardware -hardware /home/builder/.arduino15/packages -tools /home/builder/opt/arduino-builder/tools -tools /home/builder/.arduino15/packages -built-in-libraries /home/builder/opt/libraries/latest -logger humantags -fqbn esp8266:esp8266:generic:CrystalFreq=26,FlashFreq=40,FlashMode=dout,ResetMethod=none,baud=115200,dbg=Disabled,eesz=1M,exception=disabled,ip=lm2f,led=2,lvl=None____,vt=flash,wipe=none,xtal=80 -build-cache /tmp -build-path /tmp/451904423/build -verbose -prefs runtime.tools.i586-poky-linux-uclibc.path=/home/builder/.arduino15/packages/Intel/tools/i586-poky-linux-uclibc/1.6.2+1.0 -prefs runtime.tools.CMSIS-Atmel.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs runtime.tools.x86-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/x86-linux-gcc/7.2.0 -prefs runtime.tools.nrf5x-cl-tools.path=/home/builder/.arduino15/packages/arduino/tools/nrf5x-cl-tools/9.3.1 -prefs runtime.tools.arduinoOTA.path=/home/builder/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs runtime.tools.openocd.path=/home/builder/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs runtime.tools.dfu-util.path=/home/builder/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs runtime.tools.bossac.path=/home/builder/.arduino15/packages/industruino/tools/bossac/1.7.0-industruino -prefs runtime.tools.arm-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-linux-gcc/4.9.3 -prefs runtime.tools.linuxuploader.path=/home/builder/.arduino15/packages/arduino/tools/linuxuploader/1.5.1 -prefs runtime.tools.avr-gcc.path=/home/builder/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs runtime.tools.arc-elf32.path=/home/builder/.arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs runtime.tools.arduino101load.path=/home/builder/.arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs runtime.tools.core2-32-poky-linux.path=/home/builder/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0 -prefs runtime.tools.flashpack.path=/home/builder/.arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs runtime.tools.sketchUploader.path=/home/builder/.arduino15/packages/Intel/tools/sketchUploader/1.6.2+1.0 -prefs runtime.tools.avrdude.path=/home/builder/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs runtime.tools.CMSIS.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs runtime.tools.arm-none-eabi-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs runtime.tools.mkspiffs.path=/home/builder/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs runtime.tools.xtensa-lx106-elf-gcc.path=/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs runtime.tools.esptool.path=/home/builder/.arduino15/packages/esp8266/tools/esptool/2.5.0-3-20ed2b9 -libraries /tmp/451904423/custom -libraries /tmp/451904423/pinned /tmp/451904423/ArduinoJsonTest

Using board 'generic' from platform in folder: /home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0

Using core 'esp8266' from platform in folder: /home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0

Detecting libraries used...

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip2/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/451904423/build/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10611 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266 -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/variants/generic /tmp/451904423/build/sketch/ArduinoJsonTest.ino.cpp -o /dev/null

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip2/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/451904423/build/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10611 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266 -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/variants/generic -I/home/builder/opt/libraries/latest/arduinojson_6_14_1/src /tmp/451904423/build/sketch/ArduinoJsonTest.ino.cpp -o /dev/null

Generating function prototypes...

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip2/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/451904423/build/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10611 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266 -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/variants/generic -I/home/builder/opt/libraries/latest/arduinojson_6_14_1/src /tmp/451904423/build/sketch/ArduinoJsonTest.ino.cpp -o /tmp/451904423/build/preproc/ctags_target_for_gcc_minus_e.cpp

/home/builder/opt/arduino-builder/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/451904423/build/preproc/ctags_target_for_gcc_minus_e.cpp

Compiling sketch...

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip2/include -I/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/451904423/build/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=53 [... cut ...]

/home/builder/opt/libraries/latest/arduinojson_6_14_1/src/ArduinoJson/Object/ObjectShortcuts.hpp:44:3: note: template argument deduction/substitution failed:

/tmp/451904423/ArduinoJsonTest/ArduinoJsonTest.ino:5:21: note: mismatched types 'TChar*' and 'ArduinoJson6141_0000010::String (ArduinoJson6141_0000010::Pair::*)()const'

/tmp/451904423/ArduinoJsonTest/ArduinoJsonTest.ino:5:21: note: could not resolve address from overloaded function 'kvp.ArduinoJson6141_0000010::Pair::key'

exit status 1

bblanchon, 2020-02-24:
Hi @nsbawden,

I wrote a new page for version 6.
See [How to merge JSON objects?](https://arduinojson.org/v6/how-to/merge-json-objects/).

Best regards,
Benoit

nsbawden, 2020-02-24:
The new version compiles. I see that we now use function calls instead of references for all object iterations in version 6. Thanks for the update!

Cheers!

closed 2020-02-24 by nsbawden

bblanchon, 2020-02-25:
You're welcome @nsbawden.
Good luck with your project!