bblanchon/ArduinoJson #1193: Single backslash in input json is not parsed

Jeroen88, 2020-02-24:
If a char[] with a single backslash in the input is parsed, it disappears from the serialized de-serialized object. If a double backslash in the input is parsed, both backslashes appear in the object.

Target platform ESP32, latest Arduino IDE with gcc

MVCE:
#include <ArduinoJson.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(230400);
  Serial.println("\n\nStarted");

  DynamicJsonBuffer jsonBuffer;
  char json[] = "{\"hello\":\"w\\or\\\\ld\"}";
  JsonObject & root = jsonBuffer.parse(json);
  root.prettyPrintTo(Serial);
}

void loop() {
  // put your main code here, to run repeatedly:

}

Expected output:
{
  "hello": "w\or\\ld"
}

Given output:
{
  "hello": "wor\\ld"
}

bblanchon, 2020-02-24:
Hi,

"\o" is not a valid escape sequence in JSON.
See the hightlighting here:

```json
{"hello":"w\or\\ld"}
```

I'm pretty sure that the latest version returns `DeserializationError::InvalidInput` in this case.
I don't think it's worth fixing the old 5.x branch.

Best regards,
Benoit

Jeroen88, 2020-02-24:
Hi Bernoit, thnx for the quick reply. But in that case I would have expected the double backslash to become a single (escaped) backslash. But in that case it stays a double backslash

I am planning to use version 6 ;)

Jeroen88, 2020-02-24:
The backslashes in my issue appear differently than I enter them. I use % to illustrate my issue, but read a backslash for a %:
  char json[] = "{\"hello\":\"w%%or%%%%ld\"}";

Expected output:
{
  "hello": "w%or%%ld"
}

Given output:
{
  "hello": "wor%%ld"
}

bblanchon, 2020-02-24:
Remember that C++ escapes backslashes too.
So there are two levels of escaping, you must double every backslash in the string literal.
See a demo here: https://wandbox.org/permlink/S3YIx2srpvi1f0WY

Jeroen88, 2020-02-24:
Thnx again for your quick response! I know C++ escapes too, so I tried your example. Output now is (as shown in Preview, not in edit mode):

Started
{
  "hello": "w\\\\or\\\\\\\\ld"
}

But I can not get just one backslash in the output. If I add two backslashes (one escape character and the backslash itself) in the const char * input, the backslash has disappeared in the output. If I add four backslashes in the input, I get two backslashes in the output.

It seems impossible to get just one backslash in the output.

bblanchon, 2020-02-24:
It is impossible to get a backslash in front of an "o" because it's not a valid JSON document.

You can bypass the string escaping by using `RawJson()`, but this won't produce a valid JSON document.
See demo: https://wandbox.org/permlink/lY7WHQpClo5xKy50

Jeroen88, 2020-02-24:
I understand now that I need four backslashes in the input string to get two backslashes in the output, which is interpreted as an escaped backslash by a JSON parser. Thank you for guiding me in the right direction!

closed 2020-02-24 by Jeroen88

bblanchon, 2020-02-24:
You're welcome @Jeroen88.
Good luck with your project.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)