bblanchon/ArduinoJson #1194: Parsing directly from a Stream fails if ::available() becomes 0 halfway parsing

Jeroen88, 2020-02-24:
I have a connection with a HardwareSerial. The output from the HardwareSerial is decorated 'on the fly' to become a json string using a class derived from Stream. If I feed this derived Stream directly into jsonBuffer.parse(), the parsing fails if available() becomes 0 one (or more) times before the end of the input json. I think this is because ArduinoJson regards an available() == 0 as the end of the input. However in my situation the sender is sometimes a bit slow in sending the data, resulting in an available() of 0 (for less than 1s, the default Stream timeout). If I replace the implementation of available() to wait for the underlying stream to have data (in stead of returning an intermediate 0) the problem disappears.

I am planning to read from a remote Client (WiFiClientSecure) as well and feed the stream directly into parse(), but I reckon the same behavior will be there on small delays between frames.

My best guess is that ArduinoJson regards an available() of 0 as the end of the input, while this is not always the case. Maybe an explicit '\0' character, an EOF or a well formatted { } pair should signal the end of the data rather than just an available() of 0.

bblanchon, 2020-02-24:
Hi @Jeroen88,

ArduinoJson doesn't check `String::available()`.
It only uses `Stream::readBytes()`, which should wait for the timeout before returning.
Please try to [increase the timeout](https://arduinojson.org/v6/how-to/change-the-timeout/).

Best regards,
Benoit

Jeroen88, 2020-02-25:
@bblanchon  Thnx also for this answer! Increasing the timeout did not help yet, I am investigating further. Closing for now.

closed 2020-02-25 by Jeroen88