bblanchon/ArduinoJson #1195: Would love a deep-merge function in the ArduinoJson library

nsbawden, 2020-02-25:
Deep merging two JSON objects is very commonly done when using JSON objects as state management and cross platform communication packages. For instance, using the ArduinoJson library to pass a JSON object back and forth between an ESP webserver and a hardware controlling Arduino processor allows very simple and flexible variable and state updates on both sides without needing to change the ESP server code for every new variable added on the Arduino side. Passing partial JSON representing only changed values makes the update transfers very efficient, but requires a deep JSON merge in order to easily synchronize updates on both ends. Similar to how JQuery `extend` is so often used.

Here is my attempt at a deep merge (or extend). It appears to work in my tests. Hopefully I have covered all the possible failure (i.e. memory allocation failure) cases sufficiently. Please consider adding a deep merge like this to the ArduinoJson library. I believe it would be used a lot once people realize it's great power in JSON based state and message engines, and many other JSON uses.

Thanks! And I love ArduinoJson ... great work!

```
bool deepMergeJson(JsonObject dest, JsonObjectConst src) {
    for (auto kvp : src) {
        if (kvp.value().is<JsonObject>()) {
            if (!dest.containsKey(kvp.key())) {
                if (!dest.createNestedObject(kvp.key()))
                    return false;
            }
            if (!deepMergeJson(dest[kvp.key()], kvp.value()))
                return false;
        }
        else {
            if (!dest[kvp.key()].set(kvp.value()))
                return false;
        }
    }
    return true;
}
```

bblanchon, 2020-02-25:
Hi @nsbawden,

I updated the page [How to merge JSON objects?](https://arduinojson.org/v6/how-to/merge-json-objects/).
I also added a warning at the top because I think you may be misusing the library. Please have a look.

Best regards,
Benoit

nsbawden, 2020-02-25:
I did have memory fragmentation with other JSON libraries. I was excited to find ArduinoJson exactly because it prevents memory fragmentation. I have been monitoring the application closely and so far there are zero memory leaks or additional fragmentation and `ESP.getHeapFragmentation()` always reports 2 and `ESP.getFreeHeap()` does not increase.

What I don't know is if ArduinoJson automatically cleans up (or essentially de-fragments) it's own private memory block when it is heavily and repeatedly updated with variable length strings as JasonVariant values. My current implementation mainly changes numeric variants and that would not likely be a big issue. My merge method above should gracefully fail if ArduinoJson runs out of internal memory ... so far it has not.

Can you confirm? Does ArduinoJson clean up it's internal memory if the length of many JSON values are changed often such that they no longer fit in the same internal memory slot?

bblanchon, 2020-02-25:
No, ArduinoJson doesn't clean up or defragment.
It used to do that in version 6.6, but I removed this feature because the overhead was unacceptable.

Again, ArduinoJson is a serialization library, and you're clearly using it for something else.

You could defragment a `JsonDocument`, by making a copy, but I'm pretty sure that you want to avoid that.

Fortunately for you, replacing numbers in a JsonDocument doesn't require any memory allocation, so the size should not increase. Your only way to know is by checking the result of [JsonDocument::memoryUsage()](https://arduinojson.org/v6/api/jsondocument/memoryusage/).

nsbawden, 2020-02-25:
Good to know. For my use an occasional (probably hours to days apart) copy out, clear, then copy back (probably serialize and deserialize into/from a reserved size temp String buffer) would be acceptable as AudrinoJason is still far superior to other methods for maintaining a freely transferable , mergeable, and fully dynamic state and the copy could be scheduled during otherwise down-time.

You seem to be saying that `JsonDocument::memoryUsage()` gives end to end memory usage of the private pool, not sum total variant usage, and that is excellent as I can then use it to trigger a copy based cleanup routine when necessary. (It would be nice if this point were clear in the documentation).

I still like ArdunioJason over the alternatives, with the fact that it will never fragment malloc memory being a clear winning advantage. Careful consideration of ArduinoJason's limitations should allow me to use it for what I want without disadvantage.  Thank you for taking the time to answer my questions.

closed 2020-02-25 by nsbawden

bblanchon, 2020-02-26:
I updated [JsonDocument::memoryUsage()](https://arduinojson.org/v6/api/jsondocument/memoryusage/) to clearly state that the result include the leaked memory.

I've been thinking about a `collect()` method to trigger a garbage collection manually.
Unfortunately, I don't see a way to implement it without increasing the memory footprint.
Increasing memory usage is a breaking change, so I have to postpone it to a future major version.

nsbawden, 2020-02-26:
Cannot garbage collection be an optional switched feature, OFF by default? Then the extra memory needed during the cleanup phase would not be breaking, but only turned ON by those who needed the feature and had spare room for it. .ie. those who wish to use all your hard work of making ArduinoJson an awesome microcontroller service, but in a dynamic way instead of purely for serialization.

For my purposes, the very simple copy to another then copy back cleanup you suggested is working great and has very little extra code. On my modest state JSON object it typically takes < 100 microseconds to do the cleanup on a basic ESP8266 (test code snippet below) ... well worth it to achieve a global reusable, transferable, and easily updated JSON based state structure. :) 

```
void jsonDefrag(DynamicJsonDocument& doc) {
    Serial.print("before size "); Serial.println(doc.memoryUsage());
    int tm = micros();
    DynamicJsonDocument defrag(doc.memoryUsage());
    defrag = doc;
    doc = defrag;
    tm = micros() - tm;
    Serial.print("after size "); Serial.println(doc.memoryUsage());
    Serial.print("defrag time "); Serial.println(tm);
}
```

In fact, it is fast enough that I simply included it at the end of my merge function without bothering to tigger it by checking for memory pool fill up. Less than 100 extra microseconds has no real effect on my state controlled web server. Obviously a general feature would be triggered intelligently, probably when the pool allocator detected out of memory?

bblanchon, 2020-02-26:
The problem with each optional feature is that it requires twice more testing.
While it's not impossible, it's definitely too much work for a use case that I consider out of the scope of the library.
A manual `collect()` seems to be a good compromise.

Thank you for sharing the defrag function.
BTW, the copy constructor already calls `memoryUsage()`, so you can simplify to:

```c++
void defrag(JsonDocument& doc) {
  doc.set(DynamicJsonDocument(doc));
}
```

bblanchon, 2020-03-22:
The function `garbageCollect()` was added in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).