bblanchon/ArduinoJson #1196: [Question] Append std::vector by reference (Without copy or iterating the object) to JsonArray object.

judalvarezca, 2020-02-25:
Did I read de FAQ? yes.
Problem: I have a `std::vector<double>` buffer on one of my objects. It stores data from a sensor, and I want to get all the buffer and post it to a frontend with REST methods. Is there an easy way to append the complete object to the JsonArray, without a for? my current piece of code:

```
JsonArray array = doc->createNestedArray("data");
//Maybe something like this?
double* a = &taskParametersc1.buffer[0];
array.add(a);
```

By the look on [https://arduinojson.org/v6/api/jsonarray/add/](url) I would say no, but maybe you can put me on the right track about the Library tools. Meanwhile I'll look into docs to find a clue.

Thank you!

judalvarezca, 2020-02-25:
By the way, currently I've figured a solution, in the form:
```
for (auto j = params->buffer.begin(); j != params->buffer.end(); ++j) {
    array.add(*j);
}
```
And my question is just curiosity about the pointer/reference capabilities of the tool.

bblanchon, 2020-02-26:
Hi @judalvarezca,

There is the utility function [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/), but it doesn't natively support `std::vector`, so you need to pass a pointer and a size:

```c++
copyArray(v.data(), v.size(), array);
```

See a demo here: https://wandbox.org/permlink/ouoV9XHCSTuccEMK

I'm marking this as a feature request because [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/) should support `vector` natively.

Best regards,
Benoit

bblanchon, 2020-02-26:
PS:
I read the title of this issue again.
No, we cannot do that without making a copy. 
I understand it would be a nice optimization, but I don't currently plan on implementing that.

judalvarezca, 2020-02-26:
Oh, I see, yeah I understand, it probably will be a lot of work. Thank you for your advice :)

Regards.

closed 2020-02-26 by judalvarezca