bblanchon/ArduinoJson #1197: Enum value not serialized as number

loewekordel, 2020-02-25:
platform: esp8266
ide: vscode + pio
ArduinoJson version: 6.14.1
Espressif 8266 version: 2.3.2
Arduino core version: 2.6.3

For error logging purposes, I created an enum and tried to serialize it, but the the enum number is not serialized. Instead true or false are written to the serialized buffer?

enum:
```
typedef enum errorMap_e {
  ERRC_OK  =   0,    // No Error
  ERRC_1   =   1,    // ERROR 1
  ERRC_2   =   2,    // ERROR 2
  ERRC_3   =   3,    // ERROR 3
  ERRC_4   =   4,    // ERROR 4
  ERRC_5   =   5,    // ERROR 5
  ERRC_6   =   6,    // ERROR 6
  ERRC_7   =   7,    // ERROR 7
  ERRC_8   =   8,    // ERROR 8
  ERRC_9   =   9,    // ERROR 9
  ERRC_10  =  10,    // ERROR 10
  ERRC_100 = 100     // ERROR 100
} errorMap_t;
```
ArduinoJson definitions:
```
StaticJsonDocument<150> jsonBuffer;
DeserializationError err;
char sendbuffer[256];
```

Serialization:
```
Serial.print("time: ");
Serial.print(errList[i].time);
Serial.print("  err: ");
Serial.println(errList[i].err);

// create json
jsonBuffer["cmd"] = "log";
jsonBuffer["time"] = errList[i].time;
jsonBuffer["err"] = errList[i].err;

// Send data only, if clients are connected
if (webSocket.connectedClients()) {
  size_t len = serializeJson(jsonBuffer, sendbuffer);
  if (len == 0) {
    Serial.println("Failed to serialize json");
  }
  DEBUGMSG(sendbuffer);

  // send data
  Serial.printf("Client updated %d - log\n", num);
  webSocket.sendTXT(num, sendbuffer, len);
}
```

Serial output (sendBuffer): {"cmd":"log","time":1582671207,"err":true}
I hope the explanation is good enough.

I appreciate your effort and would be very thankful for a hint.

bblanchon, 2020-02-26:
Hi @loewekordel,

Thank you very much for reporting this issue.
This is clearly a bug.
Until I find a way to fix it, please cast the value to `int`:

```c++
jsonBuffer["err"] = (int)errList[i].err;
```

Best regards,
Benoit

closed 2020-02-26 by bblanchon

loewekordel, 2020-02-26:
Hi @bblanchon,

I appreciate the quick analysis and fix!
Is there a plan, when the next version, including the fix, will be released?

Thank you very much!

Best regards,
Georg

bblanchon, 2020-02-26:
The fix will be included in 6.15 in one or two weeks.

bblanchon, 2020-03-22:
This fix was included in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

commit 2996503b (2020-02-26): Fixed enums serialized as booleans (fixes #1197)