bblanchon/ArduinoJson #1198: strcmp and strncmp wrong

shufps, 2020-02-26:
Hi,

the bahaviour of this code is wrong - at least it was on my RISC-V code that is compiled with risc-v g++ 9.2.

The definition of strcmp states that both parameters are equal if the return value is 0 - but the return value is defined as integer and in my case, it tried to compare `addressIndex` with `auth` which gave `0x1100` back. But during the cast to `int8_t` the upper 8bit get lost and the key is accepted as equal. This probably happens during 32bit-wise comparisons that would be totally fine if not only 8 of 32bits are used to determine the result.

Better would be something like `!!strcmp(a,b)`

Probably the same problem also is true for the strncmp variant.

```
inline int8_t safe_strcmp(const char* a, const char* b) {
  if (a == b) return 0;
  if (!a) return -1;
  if (!b) return 1;
  return static_cast<int8_t>(strcmp(a, b));
}

inline int8_t safe_strncmp(const char* a, const char* b, size_t n) {
  if (a == b) return 0;
  if (!a) return -1;
  if (!b) return 1;
  return static_cast<int8_t>(strncmp(a, b, n));
}
```

bblanchon, 2020-02-26:
Hi @shufps,

Thank you very much for reporting this.
This bug was really hard to find.
Luckily, you made a complete diagnostic.
I'll fix that ASAP.

I tend to use 8-bit integers as much as I can because it reduces the size of the code on AVR.
It turned out to be a bad idea here :-(

Best Regards,
Benoit

shufps, 2020-02-26:
`This bug was really hard to find.`

Haha yes, it took me half the day because I didn't assume the bug could be in the awesome ArduinoJson-lib :sweat_smile: 

Tried to do a minimal replication on my amd64 but the problem didn't appear - so I thought it could be some memory-corruption on my µC-system or side-effects of my memory-protection or ... something else.

So I'm glad, I could find this bug :)

Thank you for fixing!

closed 2020-02-27 by bblanchon

bblanchon, 2020-03-22:
This fix was included in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

commit 375ea78f (2020-02-26): Fix #1198
commit 2641697e (2020-02-27): Fixed incorrect string comparison on some platforms (fixes #1198)