bblanchon/ArduinoJson #1199: How can I convert values in an array from a string to an integer?

UnicornsOnLSD, 2020-02-27:
I'm using Firestore to get an array and it is giving me an array of strings (data type on Firestore is set to number). Here is the array that I am using:
```
{
  "name": "projects/arduino-project-234f5/databases/(default)/documents/SensorData/Temperature",
  "fields": {
    "Temperature": {
      "arrayValue": {
        "values": [
          {
            "integerValue": "0"
          },
          {
            "integerValue": "0"
          },
          {
            "integerValue": "0"
          },
          {
            "integerValue": "0"
          },
          {
            "integerValue": "3"
          },
          {
            "integerValue": "0"
          },
          {
            "integerValue": "0"
          },
          {
            "integerValue": "25"
          },
          {
            "integerValue": "26"
          },
          {
            "integerValue": "25"
          }
        ]
      }
    }
  },
  "createTime": "2020-02-26T18:06:37.101500Z",
  "updateTime": "2020-02-27T22:04:32.247833Z"
}
```
I'm turning it into the JSON into a JsonArray like this: `JsonArray temperatureArray = doc["fields"]["Temperature"]["arrayValue"]["values"];`

How could I convert all the integerValues to integers?

bblanchon, 2020-02-28:
Hi,

Here is how you could do:

```c++
for (JsonObject temperature : temperatureArray) {
  int value = temperature["integerValue"];
  // ...
}
```

BTW, it's a very curious and inefficient way to send the values.
If I were you, I would simplify this JSON document.
Moreover, "integerValue" is a very misleading name for a string.

Best regards,
Benoit

UnicornsOnLSD, 2020-02-28:
@bblanchon Thanks for the help! Doing this worked:
```
for (JsonObject temperature : temperatureArray) {
    int value = temperature["integerValue"];
    temperature["integerValue"] = value;
}
```
About the weird JSON layout, it's not my system. It's just the way Firestore sends JSON :)

closed 2020-02-28 by UnicornsOnLSD