bblanchon/ArduinoJson #1200: Parsing Json with [] as parent and no names on elements

Rich-Hopkins, 2020-02-28:
I'm using v6 of ArduinoJson.
The json I'm trying to parse looks like this:
`[{
		"icao": "KCNI",
		"observed": "2020-02-28T15:30:00.000Z",
		"flight_category": "VFR"
	}, {
		"icao": "KGVL",
		"observed": "2020-02-28T14:53:00.000Z",
		"flight_category": "VFR"
	}
]
`
(with more elements in the array - 24 total I believe)
I have tried so many things that I'm not sure what they are now, but here's what I'm on at the moment and I have no output.
` char json[] = "[{\"icao\": \"KCNI\",\"observed\": \"2020-02-28T15:30:00.000Z\",\"flight_category\": \"VFR\"},{\"icao\": \"KGVL\",\"observed\": \"2020-02-28T14:53:00.000Z\",\"flight_category\": \"VFR\"}]";
  char json2[] = "[{1, 2}, {3, 4}]";
  // allocate the memory for the document
  StaticJsonDocument<2500> doc;
  deserializeJson(doc, json);
  JsonObject root = doc.as<JsonObject>();

Serial.println("Parsing...");
  for (JsonPair kv : root) {
    for(JsonPair pr : kv.value().as<JsonObject>()){
      Serial.println(pr.key().c_str());
      Serial.println(pr.value().as<char*>());
    }
  }`
The only output I get is my "Parsing..."

Even though it looks pretty nasty, and I realize there has to be a better way, I thought I'd try this, but alas, it doesn't yield what I'm looking for. I see the Starting array, and two Outer elements in my serial monitor, so at least I know that the outer elements are being iterated over. I don't see the Inner element message being posted, though.
` StaticJsonDocument<2500> doc;
  deserializeJson(doc, json);
  JsonArray root = doc.as<JsonArray>();

Serial.println("Starting array...");
  for (JsonVariant pair : root) {
    Serial.println("Outer element...");
    char* pairChar = pair.as<char*>();
    StaticJsonDocument<150> innerDoc;
    deserializeJson(innerDoc, pairChar);
    JsonObject pairs = innerDoc.as<JsonObject>();
    for(JsonPair kv : pairs){
      Serial.println("Inner element");
      Serial.println(kv.key().c_str());
      Serial.println(kv.value().as<char*>());
    }
  }
`

I need to figure out how to address the various elements in the list. I've tried kv[0].... and such, but can't seem to get it. Any help is appreciated. Thank you

bblanchon, 2020-02-29:
Hi @Rich-Hopkins,

To iterate over an array of objects, you can use this construct:

```c++
for (JsonObject obj : array) {}
```

In this line, the `JsonVariant` returned by the iterator is implicitly converted to a `JsonObject`.

Here is the complete code:

```c++
DynamicJsonDocument doc(2500);
deserializeJson(doc, json);

for (JsonObject obj : doc.as<JsonArray>()) {
  for(JsonPair kv : obj) {
      Serial.println(kv.key().c_str());
      Serial.println(kv.value().as<char*>());
  }
}
```

Best regards,
Benoit

Rich-Hopkins, 2020-03-02:
This works - thank you.

bblanchon, 2020-03-03:
You're welcome @Rich-Hopkins.
Good luck with your project.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)

closed 2020-03-03 by bblanchon