bblanchon/ArduinoJson #1202: Change Nested Array value

gednz, 2020-03-04:
Hi Benoit

Love your work!

I currently declare the following globally so that I can add data from various functions to `dataa`:
```
const char*  DEVICEID = "DevGW";
StaticJsonDocument<300> doca;
JsonArray devicea = doca.createNestedArray(DEVICEID); 
JsonObject dataa = devicea.createNestedObject();
```

My problem is that I want to store the `DEVICEID` value in an eeprom which I can't access before setup()
Is it possible to change the `DEVICEID` value after declaring the NestedArray?

Regards
Gerald

bblanchon, 2020-03-04:
Hi @gednz,

No, it's currently not possible to change a key or rename a member (requested in issue #979)
It wouldn't be too hard to implement, but I doubt the usefulness of this feature.

Are you sure there isn't a more straightforward solution to your problem?
Why (and how) do you need to create the `JsonDocument` before `setup()`?
It looks like you're using a global `JsonDocument`, are you aware of [the problems with a global document](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)?
Why don't you delay the creation of the document until you get all the data?

Best regards,
Benoit

gednz, 2020-03-05:
Hi Benoit

Thanks for clearing that up.

My routine was to directly collect data in various functions using `dataa["R"] = rssi;` and then when ready to transmit `serializeJson(doc, output);`. I always did a `doc.clear()` when done and never had any issues.
I can change to declaring the jsondocuments in a "sampling" function and call sub functions with return values from there. Once that completes I can then serialize to a global char before exiting.

Regards
Gerald

closed 2020-03-20 by bblanchon