bblanchon/ArduinoJson #1203: deserializeJson(doc, json):  Empty fields

Willem2015, 2020-03-05:
Hello Benoit,

Many thanks voor your very fine software and the excellent documentation!

I tried the example from https://www.instructables.com/id/ESP8266-Parsing-JSON/ and it works fine.

But with my code I have the same problem as described by deshmukhprat also 
on https://www.instructables.com/id/ESP8266-Parsing-JSON/

It sounds crazy but the first time it worked and now only empty fields are shown.
I have added a line of code to see the output (=String payload) from the website and the payload data is as expected.
I parse only a part of this data and as you can see: the parsed info is empty, please see the OUTPUT.
Please see also the attached code.
It is based on ArduinoJson version 6.14.1.
Platform ESP8266 WeMos D1 R1 with Arduino version 1.8.12.

Many thanks in advance for any tips or solution!

====================OUTPUT======================
Connecting...
Connecting...
Connecting...
payload: { "liveweer": [{"plaats": "Nieuwenhagen", "temp": "4.5", "gtemp": "1.2", "samenv": "Licht bewolkt", "lv": "78", "windr": "ZZW", "windms": "4", "winds": "3", "windk": "7.8", "windkmh": "14.4", "luchtd": "1006.8", "ldmmhg": "755", "dauwp": "1", "zicht": "45", "verw": "Vanavond buien, in het oosten eerst met onweer en hagel", "sup": "07:13", "sunder": "18:24", "image": "wolkennacht", "d0weer": "halfbewolkt", "d0tmax": "10", "d0tmin": "2", "d0windk": "3", "d0windknp": "8", "d0windms": "4", "d0windkmh": "15", "d0windr": "NO", "d0neerslag": "33", "d0zon": "30", "d1weer": "halfbewolkt", "d1tmax": "9", "d1tmin": "0", "d1windk": "2", "d1windknp": "6", "d1windms": "3", "d1windkmh": "11", "d1windr": "ZW", "d1neerslag": "50", "d1zon": "30", "d2weer": "regen", "d2tmax": "6", "d2tmin": "1", "d2windk": "2", "d2windknp": "6", "d2windms": "3", "d2windkmh": "11", "d2windr": "O", "d2neerslag": "90", "d2zon": "10", "alarm": "0"}]} 

Plaats: 
Temperatuur: 
Samenvatting: 
Luchtvochtigheid: 
Verwachting: 

* compiler output
IRAM   : 26908   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1312  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 2268  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 24984 )         - zeroed variables      (global, static) in RAM/HEAP 

De schets gebruikt 284664 bytes (27%)  programma-opslagruimte. Maximum is 1044464 bytes.
Globale variabelen gebruiken 28564 bytes (34%) van het dynamisch geheugen. Resteren 53356 bytes voor lokale variabelen. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:dc:df:b2
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 288816 bytes to 209834...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 288816 bytes (209834 compressed) at 0x00000000 in 4.8 seconds (effective 478.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Good questions get fast answers!
-->
[JSON.txt](https://github.com/bblanchon/ArduinoJson/files/4292611/JSON.txt)
[JSON.txt](https://github.com/bblanchon/ArduinoJson/files/4292623/JSON.txt)

bblanchon, 2020-03-06:
Hi @Willem2015,

Can you check the value returned by [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/)?

Best regards,
Benoit

Willem2015, 2020-03-06:
Hi Benoit,

Many thanks for your quick answer!

I have added in my code:
 DeserializationError err = deserializeJson(doc, json);
  Serial << "\n Error value returned: " << err.c_str()  ;

The serial output is:
 payload: { "liveweer": [{"plaats": "Nieuwenhagen", "temp": "7.5", "gtemp": "4.9", "samenv": "Geheel bewolkt", "lv": "79", "windr": "NW", "windms": "4", "winds": "3", "windk": "7.8", "windkmh": "14.4", "luchtd": "1000.0", "ldmmhg": "750", "dauwp": "4", "zicht": "20", "verw": "Overwegend bewolkt en lokaal lichte regen", "sup": "07:06", "sunder": "18:29", "image": "bewolkt", "d0weer": "halfbewolkt", "d0tmax": "11", "d0tmin": "2", "d0windk": "3", "d0windknp": "8", "d0windms": "4", "d0windkmh": "15", "d0windr": "NO", "d0neerslag": "8", "d0zon": "37", "d1weer": "halfbewolkt", "d1tmax": "9", "d1tmin": "3", "d1windk": "2", "d1windknp": "6", "d1windms": "3", "d1windkmh": "11", "d1windr": "W", "d1neerslag": "30", "d1zon": "40", "d2weer": "regen", "d2tmax": "10", "d2tmin": "3", "d2windk": "3", "d2windknp": "10", "d2windms": "5", "d2windkmh": "19", "d2windr": "ZW", "d2neerslag": "90", "d2zon": "10", "alarm": "0"}]}                            

 Error value returned: Ok
 Plaats: 
 Temperatuur: 
 Samenvatting: 
 Luchtvochtigheid: 
 Verwachting: 

Best regards,
Willem

Willem2015, 2020-03-06:
Hi Benoit,

Many thanks for your quick answer!

I have added in my code:

DeserializationError err = deserializeJson(doc, json);
Serial << "\n Error value returned: " << err.c_str() ;

 

The serial output is:

payload: { "liveweer": [{"plaats": "Nieuwenhagen", "temp": "7.5", "gtemp": "4.9", "samenv": "Geheel bewolkt", "lv": "79", "windr": "NW", "windms": "4", "winds": "3", "windk": "7.8", "windkmh": "14.4", "luchtd": "1000.0", "ldmmhg": "750", "dauwp": "4", "zicht": "20", "verw": "Overwegend bewolkt en lokaal lichte regen", "sup": "07:06", "sunder": "18:29", "image": "bewolkt", "d0weer": "halfbewolkt", "d0tmax": "11", "d0tmin": "2", "d0windk": "3", "d0windknp": "8", "d0windms": "4", "d0windkmh": "15", "d0windr": "NO", "d0neerslag": "8", "d0zon": "37", "d1weer": "halfbewolkt", "d1tmax": "9", "d1tmin": "3", "d1windk": "2", "d1windknp": "6", "d1windms": "3", "d1windkmh": "11", "d1windr": "W", "d1neerslag": "30", "d1zon": "40", "d2weer": "regen", "d2tmax": "10", "d2tmin": "3", "d2windk": "3", "d2windknp": "10", "d2windms": "5", "d2windkmh": "19", "d2windr": "ZW", "d2neerslag": "90", "d2zon": "10", "alarm": "0"}]}

Error value returned: Ok
Plaats:
Temperatuur:
Samenvatting:
Luchtvochtigheid:
Verwachting:

Best regards,
Willem

 

Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

bblanchon, 2020-03-07:
Hi Willem,

I saw you closed the issue.
Did you solve your problem?

Best regards,
Benoit

Willem2015, 2020-03-07:
Hello Benoit,

 

No closing the issue was a little clumsy unintended action ;-).

My problem still exists.

Sorry for the confusion.

 

As previously indicated I have added in my code:

 

DeserializationError err = deserializeJson(doc, json);
Serial << "\n Error value returned: " << err.c_str() ;

The serial output is:

payload: { "liveweer": [{"plaats": "Nieuwenhagen", "temp": "7.5", "gtemp": "4.9", "samenv": "Geheel bewolkt", "lv": "79", "windr": "NW", "windms": "4", "winds": "3", "windk": "7.8", "windkmh": "14.4", "luchtd": "1000.0", "ldmmhg": "750", "dauwp": "4", "zicht": "20", "verw": "Overwegend bewolkt en lokaal lichte regen", "sup": "07:06", "sunder": "18:29", "image": "bewolkt", "d0weer": "halfbewolkt", "d0tmax": "11", "d0tmin": "2", "d0windk": "3", "d0windknp": "8", "d0windms": "4", "d0windkmh": "15", "d0windr": "NO", "d0neerslag": "8", "d0zon": "37", "d1weer": "halfbewolkt", "d1tmax": "9", "d1tmin": "3", "d1windk": "2", "d1windknp": "6", "d1windms": "3", "d1windkmh": "11", "d1windr": "W", "d1neerslag": "30", "d1zon": "40", "d2weer": "regen", "d2tmax": "10", "d2tmin": "3", "d2windk": "3", "d2windknp": "10", "d2windms": "5", "d2windkmh": "19", "d2windr": "ZW", "d2neerslag": "90", "d2zon": "10", "alarm": "0"}]}

Error value returned: Ok

Plaats:
Temperatuur:
Samenvatting:
Luchtvochtigheid:
Verwachting:

Thanks again and a nice weekend from the Netherlands.

Best regards,
Willem

 

http://www.avg.com

bblanchon, 2020-03-07:
Hi @Willem2015,

Your program extracts the values, before calling `deserializeJson()`.
In other words, the `JsonDocument` is empty at the time you extract the values, and that's why all fields are empty.

Best regards,
Benoit

Willem2015, 2020-03-08:
Hi Benoit,

After looking to your video "_How to deserialize a JSON document with ArduinoJson 6_"  I suddenly saw my error!
I pasted the code from your Assistant completely in my code.
The line of code from the assistant  
const char* json = "{\"liveweer\":[{\"plaats\":\"Nieuwenhagen\",\"tem etc. etc......"}]}";
is meaningless and because of that the line of code from the assistant  
deserializeJson(doc, json); is also wrong.
It has to be deserializeJson(doc, **payload**); 
where payload is the String obtained earlier in my code with payload = http.getString();  

It now works as expected and I can continue with my weather station.

Many thanks for your help and of course your nice piece of software!!

bblanchon, 2020-03-09:
You're welcome Willem.
Good luck with your project.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)

closed 2020-03-09 by bblanchon