bblanchon/ArduinoJson #1205: Does the DeserializeJson affecting the serial?

ronneychu, 2020-03-07:
I am getting a JSON record from my database in this format
`{\"scheduleID\":\"1\",\"name\":\"gor\",\"time\":\"011100\",\"lid\":\"1\"}`

Before implementing deserializeJson, i check the string i pass is exactly what i want. However after implementing deserializeJson, i can't even recieve the json well, what's the problem?? 

`void RecieveOnCloud(int p1hour, int lid){

    String hour,minute,second,s2,rcv;
    getStringTime(&hour,&minute,&second);
                  
    String cmd = "AT+CIPSTART=\"TCP\",\"";
    cmd += IP;
    cmd += "\",80";
    sendAT(cmd);
    if( mySerial.find( "Error" ) )
        {
             Serial.print( "RECEIVED: Error\nExit1" );
              return;
       }else{
              Serial.print("Connected to server \r\n");
            }

      
   
      cmd = GET2 +"time="+p1hour+":00:00&lid="+lid+"\r\n";
      mySerial.print( "AT+CIPSEND=" );
      mySerial.println( cmd.length() );
      Serial.print( F("AT+CIPSEND=" ));
      Serial.println( cmd.length() );

      
    if(mySerial.find( ">" ) )
        {
        Serial.print(">");
        Serial.print(cmd);
        mySerial.print(cmd);

        bool checker = false ;
             while(mySerial.available()){

                 if(mySerial.find("IPD,")){ 
                   checker= true;}

                  if(checker== true) { 
                     String copy = mySerial.readStringUntil("\r"); 
                     rcv+= copy;
                   }
         }  

          Serial.println("result of copying:" +rcv);
                  
             s2=rcv.substring((rcv.indexOf('{')),rcv.indexOf('}')+1);
         Serial.println("result of cropping:"+s2);
           JsonToRecord(s2);

    }`
`void JsonToRecord(String s2){
  
          
        const size_t capacity = JSON_OBJECT_SIZE(4) + 60;
         DynamicJsonDocument doc(capacity);

        //const char* json = "{\"scheduleID\":\"1\",\"name\":\"gor\",\"time\":\"011100\",\"lid\":\"1\"}";
       const char* json = s2.c_str();
         Serial.println(json);  

     DeserializationError err = deserializeJson(doc, json);

      if(err){
        Serial.print("error");
        Serial.print(err.c_str());
        return;}
`

output of  commented the line of deserializeJson
[image]

output of Using the deserializeJson
[image]

bblanchon, 2020-03-07:
Hi @ronneychu,

I'm sorry, I don't understand your question.
Maybe you're looking for this:

* [Why is the input modified?](https://arduinojson.org/v6/faq/why-is-the-input-modified/)

Best regards,
Benoit

ronneychu, 2020-03-07:
i keep passing JSON string to a function call JsonToRecord in order to extract the data using a loop, Every time the same JSON(for testing). However, the extraction sometimes work sometimes not.
`void JsonToRecord(String s2){
  
          
      const size_t capacity = JSON_OBJECT_SIZE(4) + 50;
         DynamicJsonDocument doc(capacity);

       // const char* json = "{\"scheduleID\":\"1\",\"name\":\"gor\",\"time\":\"011100\",\"lid\":\"1\"}";
        const char* json = s2.c_str();
         Serial.println(json);  

     DeserializationError err = deserializeJson(doc, s2);

      if(err){
        Serial.print("error");
        Serial.print(err.c_str());
        return;}
        
        const char* ScheduleID = doc["ScheduleID"];
        const char* Time = doc["Time"];
        
        record1.mID= atoi(ScheduleID);
               Serial.print(F("ScheduleID of this record:"));  
               Serial.println(ScheduleID);  

  
             record1.time= Time;       
             Serial.print(F("record1 time:"));
             Serial.println(Time);

      return;
        
  }`
[image]

bblanchon, 2020-03-08:
Hi @ronneychu,

I'm sorry, I don't see what's wrong.
Can you create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?
What kind of hardware do you use?

Best regards,
Benoit

ronneychu, 2020-03-09:
hi, i fixed the previous problem. However, i am looping the deserializeJson in order to update the info, it's okay to extract the data for the first time, but since the second time , the deserializeJson  appears noMemory error, what is wrong? do I have to and how to reallocate DynamicJsonDocument memory space?

bblanchon, 2020-03-10:
Hi @ronneychu,

I answered on the following page:

* [How to reuse a JsonDocument](https://arduinojson.org/v6/how-to/reuse-a-json-document/)

Let me know if you have any questions or remarks.

Best regards,
Benoit

closed 2020-03-20 by bblanchon