bblanchon/ArduinoJson #1207: deserializeJson results in InvalidInput

superfloh247, 2020-03-07:
Hi,

ArduinoJson 6.14.1
ESP32 toochain 1.0.4

I'm trying to parse JSON, but it fails with InvalidInput

Here's the code:
    HTTPClient http;
    http.useHTTP10(true);
    http.begin(url);
    Serial.println("[HTTP] GET...\n");
    int httpCode = http.GET();
    Serial.print("[HTTP] GET... code: ");
    Serial.println(httpCode);
    if(httpCode == HTTP_CODE_OK) {
        DynamicJsonDocument doc(2048);
        const String json = http.getString();
        Serial.println(json);
        DeserializationError error = deserializeJson(doc, json);
        if (error) {
            Serial.print("deserializeJson() failed: ");
            Serial.println(error.c_str());
        }
        else {
            
        }
    }
    else {
        Serial.print("[HTTP] GET... failed, error: ");
        Serial.println(http.errorToString(httpCode).c_str());
    }
    http.end();

And here's serial output:
[HTTP] GET... code: 200
﻿{"charging":false,"driving":false,"online":true,"sleeping":false,"speed":0,"power":0,"odometer":69406.6199526852,"ideal_battery_range_km":242.94703638733768,"outside_temp":3,"battery_level":65,"charger_voltage":0,"charger_phases":0,"charger_actual_current":0,"charge_energy_added":0,"charger_power":0,"car_version":"2020.4.1 4a4ad401858f","trip_start":"21:26","trip_max_speed":0,"trip_max_power":0,"trip_duration_sec":3,"trip_kwh":0,"trip_avg_kwh":0,"trip_distance":-0.048460659512784332,"ts":"2020-03-07T21:26:59","latitude":52.446167,"longitude":13.358044,"charge_limit_soc":90,"inside_temperature":7.6,"battery_heater":false,"is_preconditioning":false,"sentry_mode":false}
deserializeJson() failed: InvalidInput

I already ran this JSON through https://jsonformatter.curiousconcept.com and the site reported it as valid.

Please help me with this issue.

Greetings,
Florian

bblanchon, 2020-03-08:
Hi Florian,

As you can see by clicking the link below, ArduinoJson has no problem with this JSON document:
https://wandbox.org/permlink/6WrwrNxwrOr0of3N

However, while copy-pasting the code from your message, I saw that it is prefixed by the following invisible characters: `EF BB BF`.
These three extra bytes at the beginning of a document constitute the [Byte-order mark](https://en.wikipedia.org/wiki/Byte_order_mark) for UTF-8.

`deserializeJson()` doesn't skip the byte-order mark, and I'm don't know if this is the right thing to do.
**If anyone has an opinion on this topic, please comment on this thread.**

However, you can easily detect this situation and skip the three first bytes of the string.
If you can modify the code on the server, it's probably best to remove the BOM.

Best regards,
Benoit

superfloh247, 2020-03-11:
Hi,

this works well for me:
DynamicJsonDocument doc(capacity);
        String payload = http.getString();
        if (payload.length() > 3) { // remove UTF8 BOM
            if (payload[0] == char(0xEF) && payload[1] == char(0xBB) && payload[2] == char(0xBF)) {
                Serial.println("remove BOM from JSON");
                payload = payload.substring(3);
            }
        }
        DeserializationError error = deserializeJson(doc, payload);

In my opinion you could add that as standard feature, similar to JSON comments:
#define ARDUINOJSON_ENABLE_COMMENTS 1
#define ARDUINOJSON_REMOVE_UTF8_BOM 1

Issue solved for me, thanks for the quick and on point answer!

closed 2020-03-11 by superfloh247

bblanchon, 2020-03-12:
That's a good idea; I'll make a note.