bblanchon/ArduinoJson #1208: Crash When Serializing or Parsing

lbussy, 2020-03-08:
Hello,

I was hoping you could put a pair of eyes on this and tell me where I am going wrong.  Since the error is so pervasive and nearly instant, I am sure I am doing something very basic wrong.  That means you've probably seen it 1000 times and can do this in your sleep.

I am parsing/serializing in a singleton class, and saving to SPIFFS.  As soon as I touch one of the objects, boom.  Here's the JSON:

```
{
  "apconfig": {
    "ssid": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "appwd": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
  },
  "hostname": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
  "bubbleconfig": {
    "name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "tempinf": true
  },
  "calibrate": {
    "room": 999.999,
    "vessel": 999.999
  },
  "targetconfig": {
    "targeturl": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "freq": 3600
  },
  "bfconfig": {
    "bfkey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "freq": 3600
  },
  "dospiffs1": false,
  "dospiffs2": false,
  "didupdate": false
}
```

And here's the class, jsonconfig.h:
``` c++
#ifndef _JSONCONFIG_H
#define _JSONCONFIG_H

#include "config.h"
#include <FS.h>
#include <ArduinoJson.h>
#include <ArduinoLog.h>

class JsonConfig {
    private:
        // Singleton Declarations
        JsonConfig() {} 
        static JsonConfig *single;
        // Other Declarations
        const char * filename = "/config.json";

    public:
        // Singleton Declarations
        ~JsonConfig() {single = NULL;}
        static JsonConfig* getInstance();

        // Methods
        bool parse();
        bool save();

        // Other Declarations
        char ssid[33];              // AP SSID
        char appwd[65];             // AP Pwd
        char hostname[33];          // Hostname
        char bubname[33];           // This device
        char targeturl[129];        // Target host
        char bfkey[65];             // Brewer's Friend key
        bool tempinf;               // Is temp in F
[... 204 lines cut ...]
    // Open file for writing
    File file = SPIFFS.open(filename, "w");
    if (!file) {
        Log.error(F("Failed to open configuration file." CR));
        return false;
    } else {
        // Serialize the JSON object to the file
        bool success = serializeJson(doc, file);
        // This may fail if the JSON is invalid
        if (!success) {
            Log.error(F("Failed to serialize configuration." CR));
            return false;
        } else {
            Log.verbose(F("Saved configuration as %s." CR), filename);
            return true;
        }
    }
}
```

I instantiate this from main with the following and can cause the crash either serializing or deserializing:

`JsonConfig __attribute__((unused)) *config = JsonConfig::getInstance();`

Thank you for taking a look!

bblanchon, 2020-03-10:
Hi @lbussy,

At first sight, I cannot spot a big mistake.

However, I'm very surprised that you use `single->` in every member function as if they were static methods. I could not make sense of this; what is the rationale here?

What is the problem exactly?
You say "boom"; do you mean an exception? a reboot?
If it's an exception, please use the [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to get more information.

Also, I would really help to know the exact location of the error.
Please insert some traces with [ArduinoTrace](https://github.com/bblanchon/ArduinoTrace).

Best regards,
Benoit

lbussy, 2020-03-10:
Hi Benoît, thanks for the over the shoulder look.

Yes, an exception.  I realize I need to do a little more now to find out why. I was just hoping there was a glaring mistake and one fix would do it for me. :)

As to your question about a static method - this is the form I use for a singleton class which may or may not be "classic" but it works for me so far (well, except for this.)

I'm playing with the examples (using structures) in your book.  Having an issue there too which mimics an issue I am having with my project.  I'll deconstruct it and try to figure that out - if I continue to have an issue I'll post a separate one.

Closing this as my quesiton was more a hail-mary pass than an issue.  Thanks again.

closed 2020-03-10 by lbussy

bblanchon, 2020-03-10:
You're welcome, @lbussy.
Don't hesitate to open a new issue if you're stuck.