bblanchon/ArduinoJson #1209: Missing rule to make target 'install'

Legendacan, 2020-03-09:
Hi, I've wanted to use [cget](https://cget.readthedocs.io/en/latest/index.html) to install your package to mine project  
```
cget install bblanchon/ArduinoJson
```
Everything builds correctly but fails when exporting because it is missing install command for target it fails with this
```
make: *** No rule to make target 'install'.  Stop.
Command failed: ['/usr/bin/cmake', '--build', '/home/sajmon/Documents/projects/common/cget/cget/build/tmp-16109d5a253848be8b41c00b9603c06a/build', '--config', 'Release', '--target', 'install', '--', '-j', '12']
Failed to build package bblanchon/ArduinoJson
```

I've found you have `install` commands for tests, but not for package. 

Can you add `install` to top-level cmake file please?

bblanchon, 2020-03-10:
Hi @Legendacan,

I didn't know about cget; it looks like an excellent tool.
Of course, I'll add an "install" target.

I'm a bit overwhelmed these days, so I could use a little help...

Best regards,
Benoit

Legendacan, 2020-03-11:
Hi @bblanchon,

Thank you. Cget looks promising as package manager. I would say, so far best one I've found. 

I'm starting to use your library, after I get some feel for it and have some spare time, I'd gladly help. 

Best regards,
David

closed 2020-04-07 by bblanchon

bblanchon, 2020-04-08:
Shipped in [ArduinoJson 6.15.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.15.1)

commit 1791dccb (2020-04-07): Added CMake "install" target (closes #1209)